Haskell-Things/ImplicitCAD #152: No support for recursive modules

yminsky, 2015-10-17:
I can't seem to make recursive modules work, despite some indications that it should (http://www.thingiverse.com/thing:18054).  For example, this:

    module m(x) {
        if (x > 0) { m(x-1); }
    }
    m(3);

fails to compile with this message:

    $ extopenscad  z.scad
    Loading File.
    Processing File.
    At 2: Module m not in scope.
    No objects to render

cosmicaug, 2016-07-23:
Forget recursion. It looks to me like it's simply the case that user defined modules do not work.

For instance, this:
```
module bigSphere() {
    sphere(d=100);
}

module littleSphere() {
    sphere(d=10);
}

bigSphere();
translate([150,0,0]) littleSphere();
```

produces this output:
```
$ extopenscad  spheres.scad
No objects to render
```

The Wiki at https://github.com/colah/ImplicitCAD/wiki does mention modules but does so rather cryptically:

> ## **modules**
> 
> **-** with `model` (to be explained)

It would appear as if user defined modules are to be meant to implemented at some point but not in the same way as the way that they are implemented in OpenSCAD.

julialongtin, 2017-03-07:
Um, i use them all the time. your spheres are using d= for diameter, in implicitcad, we use r= for radius.

julialongtin, 2019-09-02:
Reproduced the original issue.

nmz787, 2020-02-24:
seems diameter works these days, but recursion still doesn't. Here's a slightly more interesting example than the original:
```
module mySphere(diam, i) {
    translate([(diam/2)*i,0,0]) sphere(d=diam);
    if (i>0){
    	mySphere(diam, i-1);
    }
}

mySphere(100, 1);
```

wbrickner, 2020-06-03:
This is a really important language feature missing from ImplicitCAD.  

I'd like to be able to take advantage of recursion in my modules, but can't.  I tried a bounce-style recursion (e.g. `recursiveA` calls `recursiveB`, and `recursiveB` calls `recursiveA`), but this still doesn't work after one level, because one must be declared before the other.

Is there anything supported like a header / declaration in C? Like I could declare the existence of a module so an earlier module can depend on it, but only implement it later on?

Thanks.

wbrickner, 2020-06-03:
...or support for stack-like structures to implement my own hacky recursion iteratively?

julialongtin, 2020-06-05:
I'll try to give this a stab this weekend. thanks for the reminder. :)

julialongtin, 2020-06-07:
Fixed in https://github.com/colah/ImplicitCAD/pull/266 .

closed 2020-06-07 by julialongtin