Haskell-Things/ImplicitCAD #154: Slow responce from www.implicitcad.org

firegurafiku, 2015-11-11:
The site is running, but responds so slowly! I've noticed that it's running on a Heroku instance, I guess it's to make online preview possible. However, online preview is not functional now and the rest of site is effectively static. Is it a good idea to move static content to GithubPages and dedicate Heroku instance resources for preview rendering only when it's fixed?

I can do this if I have the site sources.

julialongtin, 2017-03-06:
I've built a new host for this, that uses a 4 core 8 gig of ram machine for the web server, backed by a 32 core 256 gig ram host for rendering. We're in the process of moving now.

julialongtin, 2017-03-13:
Services have been moved. \o/!

closed 2017-03-13 by julialongtin