Haskell-Things/ImplicitCAD #155: Q: Blackboard letters ℝ, ℕ and ℤ

firegurafiku, 2015-11-11:
I've just started to look through the source code and found the following fragment in `Definitions.hs`.

```
-- Let's make things a bit nicer. 
-- Following math notation ℝ, ℝ², ℝ³...
type ℝ = Float
type ℝ2 = (ℝ,ℝ)
type ℝ3 = (ℝ,ℝ,ℝ)

type ℕ = Int
```

It's obvious that the original author liked APL a lot, and had a typographic keyboard layout too, but...
* Why not `ℝ²` and `ℝ³`, with superscript numbers?
* It's common to denote the *natural* numbers set by `ℕ`, not integer. Wouldn't `ℤ` suit better?

mmachenry, 2015-11-30:
If anything I'd probably move in a direction away from these non-keyboard keys because they're really inconvenient for actually working with the library. I tend to write my objects directly in Haskell as it's a more powerful language than OpenSCAD and needing to annotate my types with subscript would make these even harder.

bgamari, 2016-08-03:
I agree with @mmachenry. I think the rather unique style used in this project hurts it more than it helps. I've been long tempted to submit a pull request de-tabbing the codebase and reverting things to standard Haskell names.

julialongtin, 2016-11-12:
I actually have learned to like it. ;)

julialongtin, 2017-01-19:
I'm going to close this, as no-one wants to argue with me. ;)

closed 2017-03-07 by julialongtin