Haskell-Things/ImplicitCAD #160: implicit-0.0.5 depends on MemoTrie-0.6.4 which failed to install.

Pupo1, 2015-12-25:
I get an error during the install of implicit on Linux Mint 17.1 Rebecca. 

**
$ cabal install implicit
Resolving dependencies...
Configuring MemoTrie-0.6.4...
Building MemoTrie-0.6.4...
Preprocessing library MemoTrie-0.6.4...

src/Data/MemoTrie.hs:3:51: Unsupported extension: EmptyCase
Failed to install MemoTrie-0.6.4
cabal: Error: some packages failed to install:
MemoTrie-0.6.4 failed during the building phase. The exception was:
ExitFailure 1
implicit-0.0.5 depends on MemoTrie-0.6.4 which failed to install.
vector-space-0.10.2 depends on MemoTrie-0.6.4 which failed to install.
$**

l29ah, 2015-12-31:
Maybe you want to upgrade your GHC.

Pupo1, 2016-01-02:
$ghc --version
The Glorious Glasgow Haskell Compilation System, version 7.6.3

but still I get exactly the same error message...

l29ah, 2016-01-02:
On Fri, Jan 01, 2016 at 05:27:26PM -0800, Pupo1 wrote:
> $ghc --version
> The Glorious Glasgow Haskell Compilation System, version 7.6.3
> 
> but still I get exactly the same error message...

7.10.2 here, it compiles fine.

-- 
()  ascii ribbon campaign - against html mail
/\  http://arc.pasp.de/   - against proprietary attachments

closed 2016-01-02 by Pupo1