Haskell-Things/ImplicitCAD #166: instructions do not work for OSX

tcurdt, 2016-02-08:
> Try it!
> extopenscad test:
> Make a test file: echo "circle(30);" > test.escad
> Run extopencad: extopenscad test.escad

After installing through homebrew and `cabal update && cabal install implicit` there is no `extopenscad`.

firegurafiku, 2016-02-08:
These don't work for Fedora either. Try `~/.cabal/bin/extopenscad` instead.

elenzil, 2016-06-24:
there's no `~/.cabal/bin` at all.
aha, looks like it installed into ~/Library/Haskell/bin .

julialongtin, 2017-03-07:
Does one of you using OS/X mind refining the directions for me? I am not a Mac (or windows) user.

LiamDGray, 2017-06-25:
On El Capitan with Homebrew version 1.2.3, **cabal install implicit** puts **extopenscad** into ~/.cabal/bin

rvt, 2019-03-19:
my result:

```
bash-3.2$ pwd
/Users/rvt/.cabal/bin
bash-3.2$ ls -ial
total 38416
55793868 drwxr-xr-x   7 rvt  staff      238 Mar 19 21:15 .
55789573 drwxr-xr-x  10 rvt  staff      340 Mar 19 21:15 ..
55808979 -rwxr-xr-x   1 rvt  staff  5097696 Mar 19 21:15 Benchmark
55808410 -rwxr-xr-x   1 rvt  staff  3421856 Mar 19 21:13 criterion-report
55808981 -rwxr-xr-x   1 rvt  staff  4522588 Mar 19 21:15 extopenscad
55793871 -rwxr-xr-x   1 rvt  staff  1377536 Mar 19 20:46 hspec-discover
55808983 -rwxr-xr-x   1 rvt  staff  5238568 Mar 19 21:15 implicitsnap
```

seems to be installed fine on OSX : 10.11.6

closed 2020-01-27 by julialongtin