Haskell-Things/ImplicitCAD #167: Floats without leading zero lead to parse errors

airbornemint, 2016-02-20:
Parses:

    translate([0.1,-0.1,0.1]) sphere(0.5);

Doesn't parse:

    translate([.1,-0.1,0.1]) sphere(0.5);

Also doesn't parse:

    translate([0.1,-.1,0.1]) sphere(0.5);

firegurafiku, 2016-02-21:
Haskell doesn't parse numbers like those, either.
I could never understand, why so many languages support omitting leading zero before the dot, and what is practical about it.

airbornemint, 2016-02-22:
My guess is that other languages support that because that's how people write numbers in real life. So, as far as OpenSCAD is concerned, it comes down to being more like Haskell or more like what people with no Haskell background expect.

firegurafiku, 2016-02-22:
> My guess is that other languages support that because that's how people write numbers in real life.

Please, don't be so anglocentric. That form is not that common in the rest of world, and gets misleading sometimes. I personally really like that Haskell bans those `.1` and `1.` shorthands.

> So, as far as OpenSCAD is concerned, [...]

That's the stronger argument. But I'd treat this as a minor incompatibility, which would make parser a bit more complex. Given the fact that the project isn't actively maintained now, I don't thing it's ever going to be implemented by someone but you.

Anyway, it's not me who is making decisions here. Let's ask @mmachenry instead.

mmachenry, 2016-04-11:
I personally lean toward disallowing omission of the leading zero, like ImplicitCAD and Haskell currently do. I'm open to arguments for compatibility, but personally, I think compatibility is a lot more import for exporting ImplicitCAD to OpenSCAD rather than the other way around at the moment. So being more restrictive is not an issue.

Cookshak, 2016-07-23:
I've written a new expression parser which recognizes this number syntax, among other compatibility fixes. My goal is to correctly parse 100% of correct OpenSCAD files, and to support Extopenscad extensions. The benefit of parsing this outweighs the cost of allowing it. However, it shouldn't be too difficult to take a flag on the command line that controls language compatibility level.

l29ah, 2018-05-26:
@Cookshak, so where is it?

dimitre, 2019-04-26:
+1 here for this one

closed 2019-07-18 by julialongtin