Haskell-Things/ImplicitCAD #168: documentation doesn't say how to assign objects to variable names and copy them

sudoman, 2016-03-01:
... or describe whatever functionality is comparable in the implicitcad system. something like this is necessary to compose the code into objects or functions and for code reuse.

an advanced example in the examples/ directory showing this and other features would also be nice.

as long as i've created the issue: what is the correct way to do this in implicitcad?

sudoman, 2016-03-01:
(to be clear, i'm referring to the .escad format.)

julialongtin, 2017-03-07:
simple answer: I don't know!

I'm looking into it...

julialongtin, 2019-07-27:
ok, i have an answer, but i don't think you'll like it. :)

It's not possible to assign an object to a variable.

It is, however, possible to assign a module to a variable, and that module can have no arguments, and produce the object you're looking for.