Haskell-Things/ImplicitCAD #169: Language of extopenscad?

aggregate, 2016-03-26:
I've been using OpenSCAD for several years now and have gotten around many of the awkward aspects, but extopenscad is very appealing. The problem is that it is a very ill-specified "extended subset" of OpenSCAD with many things missing, many added, and many just working differently. For example, there is no hull() nor minkowski() -- two of the real workhorses in OpenSCAD -- and newer things, like specifying d=100 instead of r=50 in a cylinder also are not there. Semantic differences include things like setting $fn=6 globally having no impact on cylinders. I don't have a huge problem with things being different, but I have a big problem with the lack of a specification of the very-inappropriately-named extopenscad language. Is there a real, stable, language spec?

I don't have a huge problem with this language just being similar enough to OpenSCAD to make it easy for users to migrate designs, but it isn't really usable without a spec of what's there and how it works... especially given the lack of decent syntax error handling.

julialongtin, 2016-04-20:
This is something I'm working on in my 3D printing classes. We're working at http://fosscar.faikvm.com/trac/wiki/ImplicitCad

TLC123, 2016-10-24:
Is hull operations possible for signed scalar fields? I'm only familiar with it as a polyhederon operation.

Renha, 2017-11-13:
@TLC123 why should it be user's concern?

TLC123, 2017-11-13:
@Renha If Convex Hull operation  can be demonstrated to be impossible/impractical  to support by the underlying mathematical representation then user that absolutely need that functionality may abandon hope of its implementation and move on to other platforms that do .

scls19fr, 2018-02-17:
This is exactly what I will do... I new to programmatic CAD and wanted to make an enclosure.
I've found https://www.thingiverse.com/thing:155001/#files
it works fine with OpenSCAD but I like the idea of doing this inside a browser...
so I've just try ImplicitCAD.
I did so changed to original script see https://gist.github.com/scls19fr/5200c35522cd1304bf196791cd53cda6 but I'm now totally stuck up
with hull operation.
That's sad!

julialongtin, 2020-05-14:
There's been some work on language compatibility drom both ends... but no-one has solved the hull / minowski problem, as of yet.

mmirate, 2021-03-28:
Here's an interesting non-solution for the hull problem:

Given some implicit-functions fᵢ(x,y,z) which represent some shapes to be convex-hulled, if one could efficiently obtain closed-forms of the partial derivatives of each function, ∂fᵢ(x,y,z)/∂x, ∂fᵢ(x,y,z)/∂y and ∂fᵢ(x,y,z)/∂z, then I suspect that their convex hull could be defined as the following pieces:

h(x,y,z) = -1 if ∀i,j [signum(∂fᵢ(x,y,z)/∂x) = signum(∂fⱼ(x,y,z)/∂x) and signum(∂fᵢ(x,y,z)/∂y) = signum(∂fⱼ(x,y,z)/∂y) and signum(∂fᵢ(x,y,z)/∂z) = signum(∂fⱼ(x,y,z)/∂z)]
h(x,y,z) = 1 otherwise 

However, while the output does preserve correctness-of-signum under min/max CSG operations with other shapes, it only provides an indication of whether the point is inside the convex hull or not, without any indication of distance from the nearest point of the hull's boundary - so it completely breaks ray-marching, repeated applications of itself, etc.

This also would produce incorrect results if any of the component functions represented a shape which is not, itself, convex.

Lastly, even if min/max were replaced by softmin/softmax, I doubt that all of implicitcad's object repertoire has closed forms for its partial derivatives.