Haskell-Things/ImplicitCAD #170: Compiling examples produces "broken" geometry

pulsar256, 2016-04-20:
Affects 7d334a621810a8c55bbc7a17a343a6fc1c6e74d9 and the "stable" (0.0.5) release as available via cabal:

```
~  ᐅ cat test.escad
linear_extrude(height(x,y) = 20 + (y^2-x^2)/60)
  circle(30);
~  ᐅ extopenscad test.escad -o test.stl
Loading File.
Processing File.
Rendering 3D object to test.stl
With resolution 2.4021830874975945
In box ((-30.0,-30.0,0.0),(30.0,30.0,41.0))
ExtrudeRM 0.0 Nothing Nothing Nothing (Circle 30.0) (Right <function>)
~  ᐅ meshlab test.stl > /dev/null 2>&1 
```
[image]

Tried different resolutions and output types (obj), results were all the same.

julialongtin, 2016-04-20:
Can you try the version at http://kalli1.faikvm.com/ImplicitCAD/Stable ?

pulsar256, 2016-04-21:
@julialongtin same results
```
~/development/Implicit_Stable (master ✔) ᐅ git rev-parse HEAD
b9ddf26c6e9a6104a831bb092c35afb02c84013d
~/development/Implicit_Stable (master ✔) ᐅ ls -lah $(which extopenscad)
-rwxr-xr-x 1 pulsar users 12M 21. Apr 10:14 /home/pulsar/.cabal/bin/extopenscad
```

ochafik, 2017-01-22:
Is there any update on this? Bumping into the same issue as I'm running ImplicitCAD for the first time...

julialongtin, 2017-03-07:
I'm still trying to hunt this down. more simple examples would be helpful.

wbrickner, 2017-04-11:
Anything? A year later?

l29ah, 2018-05-25:
Reproduced it on c597ee39074120a969b99404ba7ff56b598c6d4f. I expected changing ℝ to Float will affect it, but it haven't.

julialongtin, 2019-10-16:
Fixed, at least temporarily. This is the same bug as #98 . The buggy optimization is commented out. Will close this issue, and leave #98 open, as a tracker to ensure the bug is fixed.

closed 2019-10-16 by julialongtin

pulsar256, 2019-10-26:
Could you maybe point me to the commit with the workaround? Curious to see what fixed that. Maybe this is is something to learn from :)

julialongtin, 2019-10-28:
https://github.com/colah/ImplicitCAD/commit/6e51e5f263d2cdedbdd230c107fd4c096f913f46

If you can figgure out what's wrong with the math, let me know! :)

pulsar256, 2019-10-28:
beats me, maybe it does affect gcc optimization in a way? Which makes me wonder if that could have been related to the issue. Seems there are a couple of lags which would sacrifice precision over performance (https://wiki.haskell.org/Performance/Floating_point) - but on the other hand, i would expect "rounding errors" to have a different manifestation.

julialongtin, 2019-10-28:
yeah, i'll probably take a whack at this during a performance release.. until then...