Haskell-Things/ImplicitCAD #172: parse error when the last list is a comment

sjoerdtimmer, 2016-05-06:
The parser "expects" an extra "\n" when the file ends with a line that starts with "//"

firegurafiku, 2016-05-06:
This particular behaviour may be thought not only as a bug, but [as a feature][1], too. To solve this issue we may just make the final newline mandatory and clearly state that in documentation (which isn't here currently).

[1]: http://stackoverflow.com/questions/729692/why-should-text-files-end-with-a-newline

sjoerdtimmer, 2016-05-06:
Lines should definity end with \n.
However, the parser somehow ignores this \n if the line starts with "//"
(i.e. somehow this newline character is considered a comment too? and the parser requires a newline at the end of the file).

I think it is bad design if a newline character is part of the language (besides being considered as space between tokens).

firegurafiku, 2016-05-06:
I think, this issue may be worked around as simple, as applying the following patch:

```diff
--- a/Graphics/Implicit/ExtOpenScad/Parser/Statement.hs
+++ b/Graphics/Implicit/ExtOpenScad/Parser/Statement.hs
@@ -6,8 +6,8 @@
 import Graphics.Implicit.ExtOpenScad.Parser.Expr
 
 parseProgram :: SourceName -> [Char] -> Either ParseError [StatementI]
-parseProgram name s = parse program name s where
+parseProgram name s = parse program name (s ++ "\n") where
     program = do
         sts <- many1 computation
         eof
         return sts
```

Could you check it please?

sjoerdtimmer, 2016-05-08:
That should indeed work. I'll try it soon. Still, it would be interesting
to know why the parser needs this extra return... maybe it has other
unwanted side effects?

julialongtin, 2019-07-18:
Fixed in what is currently master... along with a lot of other parser bugs.

closed 2019-07-18 by julialongtin