Haskell-Things/ImplicitCAD #177: Incompatibility: ternary operator with comparison not recognized

Cookshak, 2016-07-16:
The parser does not recognize a ternary operator expression that has a comparison as the leftmost expression:
a < b ? 3 : 4

The workaround is to add parentheses:
(a < b) ? 3 : 4

There is a disable test in the parser test suite to check for this.

julialongtin, 2016-11-04:
Is this one of the parser bugs you've addressed?

Cookshak, 2016-11-04:
I don't think I worked how to fix this in the original parser, just in the
alternate one.

Are you just grabbing the pull request i submitted to mmachenry's repo, or
are you looking at mods from my fork? I haven't created a pull request for
most of the work I've done, yet.

The only pull request I've added to mmachenry is to fix let expressions.

julialongtin, 2016-11-05:
I'm digging into the mods in the fork, but a lot of it is above my
haskell. can you pop onto the IRC channel, #implicitcad on freenode,
so we can discuss this all?

Sorry to be a blocker for this obviously good work, but I'm trying not
to commit code I do not understand.. and there's a lot I don't
understand in ImplicitCAD. :)

Julia Longtin

julialongtin, 2019-09-01:
Integrated all of the alternate expression parser work, without breaking the functionality of the original parser. The test in the test suite runs successfully.

closed 2019-09-01 by julialongtin