Haskell-Things/ImplicitCAD #178: Feature request import polygon meshes like STL.

TLC123, 2016-08-01:
According to this paper it possible to do an exact representation of polygonal objects by differentiable scalar fields based on binary space partitioning. Correct me if i'm wrong but isn't  distance fields the internal representation in ImplicitCAD?

http://www.hyperfun.org/TRP/FPA_BSPfields_CAD11.pdf

doug-moen, 2017-02-11:
According to the paper, the algorithm is very slow when you have more than a couple hundred triangles. Check out the timings in table 2. The "Turtle" model, with 103,000 polygons, takes 4.5 minutes to convert to a BSP, and then costs 18.4 seconds per 1000 distance function evaluations. By comparison, the classic Yoda model on Thingiverse has 614,000 triangles.

TLC123, 2017-02-11:
Yeah, holding out for faster algos and hardware then. Point being conversion is doable if someone want do dive in the deep end.
One such obvious route would be to start and iterate upwards from severely reduced meshes and monitor local error levels until some acceptabel value of approximation is reached. What takes a long churning today is preview speed in five years. Similar how ray-tracing was first invented on ancient machines. Basic math is the same but now we just  dial up the output.
 
Mesh import to scalar fields will always be a trade-of between precision construction drawing, high poly figurines like Yoda or just visual reference geometry(easy).

Close or keep open for future work either way i´ll continue the quest.