Haskell-Things/ImplicitCAD #180: Compile error with GHC 7.8/7.6/7.4

hvr, 2016-11-29:
I just noticed your recent package upload of `implicit-0.1.0` was flagged by the Hackage matrix builder with the following compile failures (the 1st one is with GHC 7.8; the 2nd one occured with GHC 7.6 and appears to be due to version bounds on `bytestring` not being accurate). If you don't intend to support older GHC versions you can also just declare `base >= 4.8` (which would exclude GHC 7.8 and older).

You can fixup the meta-data yourself via

https://hackage.haskell.org/package/implicit-0.1.0/implicit.cabal/edit

or I can do it for you.

If you have any questions, please let me know!

```
Configuring component lib from implicit-0.1.0
Preprocessing library implicit-0.1.0...
[ 1 of 45] Compiling Graphics.Implicit.Export.Render.GetLoops ( Graphics/Implicit/Export/Render/GetLoops.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/Export/Render/GetLoops.o )
[ 2 of 45] Compiling Graphics.Implicit.Definitions ( Graphics/Implicit/Definitions.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/Definitions.o )
[ 3 of 45] Compiling Graphics.Implicit.MathUtil ( Graphics/Implicit/MathUtil.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/MathUtil.o )
[ 4 of 45] Compiling Graphics.Implicit.ExtOpenScad.Definitions ( Graphics/Implicit/ExtOpenScad/Definitions.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Definitions.o )

Graphics/Implicit/ExtOpenScad/Definitions.hs:51:10: Warning:
    No explicit implementation for
      ‘GHC.Base.return’
    In the instance declaration for ‘Monad ArgParser’
[ 5 of 45] Compiling Graphics.Implicit.ExtOpenScad.Parser.Util ( Graphics/Implicit/ExtOpenScad/Parser/Util.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Parser/Util.o )
[ 6 of 45] Compiling Graphics.Implicit.ExtOpenScad.Util.StateC ( Graphics/Implicit/ExtOpenScad/Util/StateC.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Util/StateC.o )
[ 7 of 45] Compiling Graphics.Implicit.ExtOpenScad.Util.OVal ( Graphics/Implicit/ExtOpenScad/Util/OVal.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Util/OVal.o )
[ 8 of 45] Compiling Graphics.Implicit.ExtOpenScad.Util.ArgParser ( Graphics/Implicit/ExtOpenScad/Util/ArgParser.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Util/ArgParser.o )
[ 9 of 45] Compiling Graphics.Implicit.ExtOpenScad.Eval.Expr ( Graphics/Implicit/ExtOpenScad/Eval/Expr.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Eval/Expr.o )
[10 of 45] Compiling Graphics.Implicit.ObjectUtil.GetImplicit2 ( Graphics/Implicit/ObjectUtil/GetImplicit2.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ObjectUtil/GetImplicit2.o )
[11 of 45] Compiling Graphics.Implicit.ObjectUtil.GetImplicit3 ( Graphics/Implicit/ObjectUtil/GetImplicit3.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ObjectUtil/GetImplicit3.o )
[12 of 45] Compiling Graphics.Implicit.ObjectUtil.GetBox2 ( Graphics/Implicit/ObjectUtil/GetBox2.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ObjectUtil/GetBox2.o )
[13 of 45] Compiling Graphics.Implicit.ObjectUtil.GetBox3 ( Graphics/Implicit/ObjectUtil/GetBox3.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ObjectUtil/GetBox3.o )
[14 of 45] Compiling Graphics.Implicit.ObjectUtil ( Graphics/Implicit/ObjectUtil.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ObjectUtil.o )
[15 of 45] Compiling Graphics.Implicit.ExtOpenScad.Parser.Expr ( Graphics/Implicit/ExtOpenScad/Parser/Expr.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Parser/Expr.o )
[16 of 45] Compiling Graphics.Implicit.ExtOpenScad.Parser.Statement ( Graphics/Implicit/ExtOpenScad/Parser/Statement.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Parser/Statement.o )
[17 of 45] Compiling Graphics.Implicit.ExtOpenScad.Eval.Statement ( Graphics/Implicit/ExtOpenScad/Eval/Statement.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/ExtOpenScad/Eval/Statement.o )
[18 of 45] Compiling Graphics.Implicit.Export.MarchingSquaresFill ( Graphics/Implicit/Export/MarchingSquaresFill.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/Export/MarchingSquaresFill.o )
[19 of 45] Compiling Graphics.Implicit.Export.Symbolic.Rebound2 ( Graphics/Implicit/Export/Symbolic/Rebound2.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/Export/Symbolic/Rebound2.o )
[20 of 45] Compiling Graphics.Implicit.Export.Symbolic.Rebound3 ( Graphics/Implicit/Export/Symbolic/Rebound3.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/Export/Symbolic/Rebound3.o )
[21 of 45] Compiling Graphics.Implicit.Export.TextBuilderUtils ( Graphics/Implicit/Export/TextBuilderUtils.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/Export/TextBuilderUtils.o )
[22 of 45] Compiling Graphics.Implicit.Export.PolylineFormats ( Graphics/Implicit/Export/PolylineFormats.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/Export/PolylineFormats.o )
[23 of 45] Compiling Graphics.Implicit.Export.TriangleMeshFormats ( Graphics/Implicit/Export/TriangleMeshFormats.hs, /tmp/matrix-worker/1480425488/dist-newstyle/build/x86_64-linux/ghc-7.8.4/implicit-0.1.0/build/Graphics/Implicit/Export/TriangleMeshFormats.o )

Graphics/Implicit/Export/TriangleMeshFormats.hs:13:55:
    Module ‘Prelude’ does not export ‘mconcat’
cabal: Failed to build implicit-0.1.0 (which is required by exe:implicitsnap
from implicit-0.1.0, exe:extopenscad from implicit-0.1.0 and others).
```

```
Configuring component lib from implicit-0.1.0
Preprocessing library implicit-0.1.0...

Graphics/Implicit/Export/TriangleMeshFormats.hs:19:18:
    Could not find module `Data.ByteString.Builder.Internal'
    It is a member of the hidden package `bytestring-builder-0.10.6.0.0'.
    Perhaps you need to add `bytestring-builder' to the build-depends in your .cabal file.
    It is a member of the hidden package `bytestring-builder-0.10.8.1.0'.
    Perhaps you need to add `bytestring-builder' to the build-depends in your .cabal file.
    It is a member of the hidden package `bytestring-0.10.8.1'.
    Perhaps you need to add `bytestring' to the build-depends in your .cabal file.
    Use -v to see a list of the files searched for.
cabal: Failed to build implicit-0.1.0 (which is required by exe:implicitsnap
from implicit-0.1.0, exe:extopenscad from implicit-0.1.0 and others).
```

julialongtin, 2016-12-09:
Where is the matrix you're talking about? I'd like to support earlier haskell versions, but do not have the infrastructure...

hvr, 2016-12-09:
@julialongtin http://matrix.hackage.haskell.org/ is what I refer to as the matrix builder; however the failure above was detected by the in-dev next-gen prototype of it.

However, you can also put the Travis CI service to use, via e.g.

https://github.com/hvr/multi-ghc-travis/blob/master/make_travis_yml_2.hs

If you need assistance with it, I can help you set it up.

julialongtin, 2017-03-11:
I've fixed this to compile against the 7.8 shipped with debian stable. Unless travis implementation is stupid-easy, I'll probably only support 7.8 on debian stable, and 8.0.x on debian unstable.

julialongtin, 2017-03-12:
Closing, as 7.8 and 8.0.x are now both supported.

closed 2017-03-12 by julialongtin