Haskell-Things/ImplicitCAD #181: Excessively long run time for simple file

MattEttus, 2016-12-17:
The attached file generates dominoes.  It renders in a couple of seconds in OpenSCAD but takes many minutes when run with:

   $ extopenscad test.scad -o test.stl 

The output it eventually produces looks good, and extopenscad does do rounded corners, so it does have to work a bit harder than OpenSCAD, but this does seem excessive.  Is there something I'm doing wrong?

[test.tar.gz](https://github.com/colah/ImplicitCAD/files/658702/test.tar.gz)

TLC123, 2016-12-17:
Seem pretty normal to me, for non optimized marching cubes.
Both polygonization and image rendering really could benefit from GPU acceleration.
There are  really fast GLSL shaders for raymarching  Signed Distance Fields available.
a SMOP to combine.

julialongtin, 2016-12-17:
This is normal, and also something I'm trying to work on, as I wrap my head about the code. GPU acceleration is currently off of the table, because of a lack of support on free-software-only stacks. I've been doing a lot of work on optimizing the code at every step.

closed 2016-12-17 by julialongtin