Haskell-Things/ImplicitCAD #198: extopenscad fails in strange place

nuclearcat, 2018-03-31:
Here is simple scad design that generate stepper motor hat and works well in openscad, but strangely fails in extopenscad (git version):

`
mount_height=22;
totalcircle=46;
wiresquare=17;

module mounts() {
    hull() {
    translate([-35/2,0,0])
        cylinder(d=8,h=2.5);
    translate([35/2,0,0])
        cylinder(d=8,h=2.5);
    }
}

module hat() {
difference () {
 translate([0,0,-2])
   cylinder(d=totalcircle,h=4);
 cylinder(d=29,h=2.5);
 translate([0,8,-2.5])
    cylinder(d=10,h=5);
  mounts();
  translate([35/2,0,-2.5])
   cylinder(d=4.2,h=3);
  translate([-35/2,0,-2.5])
   cylinder(d=4.2,h=3);
  translate([0,-wiresquare,1.5])
    cube([16,10,3], center=true);

 };
}

hat();
`

Loading File.
Processing File.
(line 14, column 8):
unexpected "h"
expecting "//", "/*" or "("

l29ah, 2018-05-24:
Try removing `;` at line 29.

ShamrockLee, 2019-01-24:
It seems to happen on programs with my code:
```
module elipse_torus(rout, rain, rbin, rthick) {
    fa_circle = 0.5.;
    rotate_extrude(convexity = 10, $fn = 0, $fa = 0.500000, $fs = 2.000000)
    multmatrix([[1, 0, 0, rout], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]])
    union() {
        for (theta=[0: fa_circle*rthick/max(rain, rbin): 360]) {
            translate([rain*cos(theta), rbin*sin(theta), 0])
            circle(r=rthick, $fa=fa_circle*4);
        }
        }
    // scale([1, rbin/rain, 1])
    // circle($fn=0, $fa=2.000000, $fs=0.500000, r=rain);
}

la = 280;
lb = 180;
d0 = 1.5;
a = 34.5;
b = 54.5;
h = 40;
elipse_torus(b, b-a, h/2, d0/2);
```
output:
```
Loading File.
Processing File.
(line 1, column 8):
unexpected "e"
expecting "//", "/*" or "("
```
What's wrong with it?

julialongtin, 2019-01-24:
Error messages in Implicitcad do not work well at highlighting the exact
place parsing failed. I don't recognise the multimatrifunction in your
module, what is it supposed to do?

ShamrockLee, 2019-01-24:
Oops! It is an OpenSCAD function which can be replace with translate() here.
```
module elipse_torus(rout, rain, rbin, rthick) {
    fa_circle = 0.5.;
    rotate_extrude($fn = 0, $fa = 0.500000, $fs = 2.000000)
    translate([rout, 0, 0])
    union() {
        for (theta=[0: fa_circle*rthick/max(rain, rbin): 360]) {
            translate([rain*cos(theta), rbin*sin(theta), 0])
            circle(r=rthick, $fa=fa_circle*4);
        }
        }
    // scale([1, rbin/rain, 1])
    // circle($fn=0, $fa=2.000000, $fs=0.500000, r=rain);
}

la = 280;
lb = 180;
d0 = 1.5;
a = 34.5;
b = 54.5;
h = 40;
elipse_torus(b, b-a, h/2, d0/2);
```
The error message does not change.

ShamrockLee, 2019-01-24:
I see it! It's just an extra decimal point. Thank you for telling me about the characteristic of the error messages!

closed 2019-03-05 by julialongtin