Haskell-Things/ImplicitCAD #201: extopenscad: trigonometric functions aren't compatible with openscad

l29ah, 2018-05-25:
In openscad they take degrees, while in implicitcad they take radians, but it is never mentioned anywhere.

julialongtin, 2018-05-30:
I'm actually good with this, but i think that a compatibility flag makes sense. similar to $RES to set the resolution, $babylonian to use 360 degree notation.

acwest, 2019-03-12:
I have written a LOT of geometry library code for OpenSCAD that would be broken by this. I like the idea of a compatibility flag, as I never really liked the way the trig functions work in degrees for OpenSCAD

julialongtin, 2019-06-29:
I'll be adding a flag for this shortly.

julialongtin, 2022-10-09:
A flag for this has been added and merged. please check if enabling the openscad compatibility flag fixes this for you.