Haskell-Things/ImplicitCAD #202: rotate() is surprisingly slow

l29ah, 2018-05-25:
```
$res = 1;
cylinder(100);
```
2.18s user 1.68s system 238% cpu 1.623 total

```
$res = 1;
rotate([0, 90, 0])
	cylinder(100);
```
393.73s user 191.56s system 232% cpu 4:11.45 total

julialongtin, 2018-05-25:
I've already narrowed this down to being a side effect of one of 8 git
commits, and expect to know what happened sometime today. understanding it
well enough to not just 'undo' it may take the weekend.

julialongtin, 2018-05-25:
76d4720 is broke.
d62d3a1f8 works.

in case you want to join the hunting. :)

julialongtin, 2018-05-25:
in fact, could you check those two revisions, to make sure i'm hunting the
same bug you are?

l29ah, 2018-05-25:
Nah, d62d3a1f8c738efd9575849b56517a56091e83cb still takes much time (and then killed by oom_kill for taking over 6GB of RAM, unlike master).

julialongtin, 2018-05-25:
OK. two separate bugs. can you ballpark some "worked on commit XXX, broken
on commit YYY", while i finish hunting the performance regression i'm
currently fighting?

Julia Longtin

l29ah, 2018-05-25:
cca173e93e3a748d82c855e22efbc958346d89b8 is the earliest i could test w/o diving into ghc incompatibilities, and it consumes all the RAM there as well.

l29ah, 2018-05-25:
I've noticed the rotation produces a suboptimal bounding box: `In box ((-173.20508075688772,-173.20508075688772,-173.20508075688772),(173.20508075688772,173.20508075688772,173.20508075688772))`.
I wonder if it affects performance in any way.

julialongtin, 2018-05-25:
oh, it sure will. increases the area that needs raytracing.
a better implementation of the bounding box for simple rotates would help.

commit bbc6ad52 (2018-06-11): Merge pull request #207 from elimohl/master

Better bounding box calculation. Fix #202
commit e7d35e81 (2018-06-11): Better bounding box calculation. Fix #202
closed 2018-06-11 by julialongtin