Haskell-Things/ImplicitCAD #204: Strange behaviour of the rounded union

elimohl, 2018-05-26:
The union of two rectangles:
[image]

I expected the rounded union be something  like
[image]

But actually I get this:
```
union(r=10) {
    square([10, 10]);
    translate([10, 0])
        square([10, 20]);
}
```
[image]

elimohl, 2018-05-26:
```
union(r=10) {
    square([10, 10]);
    translate([10, -5])
        square([10, 20]);
}
```
[image]

TLC123, 2018-05-26:
The rounding does not take in count the gradient normal of the two fields
where they meet

Renha, 2018-05-29:
@TLC123 why?

l29ah, 2019-06-28:
Also occurs with 3D objects:
[image]

l29ah, 2019-06-29:
The culprit is
```
getImplicit2 (UnionR2 r symbObjs) =
    \p -> let
        objs = map getImplicit2 symbObjs
    in
        if r == 0
        then minimum $ map ($p) objs
        else rminimum r $ map ($p) objs
```
I don't really understand how is it supposed to work though, and it feels to me that it's not enough to know the distances from a point to each of the objects to find the distance to the rounded union of the objects.

l29ah, 2019-07-09:
Reading https://christopherolah.wordpress.com/2011/11/06/manipulation-of-implicit-functions-with-an-eye-on-cad/
```
(NB. This particular formation of rounded unions is a little bit naive. If you union and the sides aren’t orthogonal, this version will make a bump, which is not the expected behavior. This can be corrected for quite easily by looking at the dot product of the discrete gradients.)
```
I have no idea how to do this w/o explosive computational complexity growth.

julialongtin, 2021-02-28:
still present.

julialongtin, 2021-06-25:
hmm. still present.