Haskell-Things/ImplicitCAD #205: How to use the implicitsnap.hs

onekk, 2018-06-08:
First, thanks for the good work.

I wanted to use the web interface locally, so i have compile implicitsnap.hs in the hope to obtain a server like those found on the site, to use it like a local IDE.

Maybe i was hoping wrong things or maybe I'm doing something wrong.

I launch the bynary implictisnap and point my webbrowser to localhost:8000 and then i obtain a page with "**fall through**"   indication and nothing else.

Best Regards

Carlo D.

julialongtin, 2018-06-08:
Implicitsnap is the backend for that web site; the web site itsself is the
pile of ruby code at http://kalli1.faikvm.com/ImplicitCAD/Website. I'm not
much of a ruby programmer, so it's not the best code base.

if you need setup help, let me know.

onekk, 2018-06-08:
Thanks, I was searching for a viable solution to obtain a similar workflow as in Openscad in Linux

Having one window for each task compile, edit and view is not a mayor annoyance, but a direct stl viewer (MeshLab is working but you need to import a drawing each time it is modified) for Linux seems to be difficult to find.

Regards

Carlo D.

julialongtin, 2018-06-08:
Patches warmly welcome. :)

closed 2018-06-25 by julialongtin