Haskell-Things/ImplicitCAD #213: Random Number Generator

deckar01, 2018-07-29:
The `rands(min, max, count)` RNG function does not currently seem to be supported. It would be nice if there was a shortcut for generating a single value also `rand(min, max)`.

wbrickner, 2020-06-08:
I would also love this, I want to do procedural modeling of organic structures.

I am clueless when it comes to Haskell, but if the maintainer/s have time I would very deeply appreciate it! API doesn't need to match that of OpenSCAD for my purposes, anything would work.

Thanks everyone for all your hard work, very valuable.

lepsa, 2022-10-09:
@julialongtin Should this be closed now that `rands` is merged?

julialongtin, 2022-10-09:
@lepsa yep! this will be in our next release, folks!

closed 2022-10-09 by julialongtin