Haskell-Things/ImplicitCAD #218: export gcode directly from ImplicitCAD

dimitre, 2019-04-15:
is it possible? I've been trying with
```
~/.cabal/bin/extopenscad -f ngc 2.escad
```
or
```
~/.cabal/bin/extopenscad -f gcode 2.escad
```

and it returns
```
Unrecognized 3D format: GCode
```
Thank you

julialongtin, 2019-04-15:
ImplicitCAD is not a slicer, but i'm going to leave this open... because
i've been working on a slicer, and will be releasing it after the next
ImplicitCAD release. stay tuned!

dimitre, 2019-04-15:
Excellent. I've just asked because I think one of the original purposes of the project was to be able to slice without triangulation phase which would result in the best raster for "inifinite resolution" files.

julialongtin, 2019-12-16:
https://github.com/julialongtin/slicer

It's horrible, but i'm slowly bringing it up to the level of badness of implicitcad.

julialongtin, 2019-12-22:
https://github.com/julialongtin/hslice

moved it, and have done a lot of work on it. not up to implicitcad standards, but i'm closing this issue.

closed 2019-12-22 by julialongtin