Haskell-Things/ImplicitCAD #220: Cummulative matrix transformation (translate, rotate) from OpenSCAD

dimitre, 2019-05-03:
I know the project isn't meant to be 100% compatible with OpenSCAD but I've noticed a very basic issue trying to run the following code in both of them
```c++
union() {
    cube(size=[10,20,5]);
    rotate(30)
    translate([0,30,0]) 
    {
        cube(size=[10,20,5]);
        rotate(30)
        translate([0,30,0]) 
        {
            cube(size=[10,20,5]);
        }
    }
}
```
and ImplicitCAD doens't seem to understand the cumulative translation and rotation
[image]

l29ah, 2019-06-29:
The result suggests that there's some problem with a bounding box calculation.

l29ah, 2019-06-29:
Minimal reproducer: `rotate(30) cube(10);`.

julialongtin, 2019-07-09:
Reproduced with the original code. investigating.

isovector, 2020-11-26:
I'm unable to reproduce in the Haskell library:

```haskell

main :: IO ()
main = writeSTL 1 "/tmp/yo.stl" $
  let cube = cubeR 0 True (10, 20, 5)
   in union
        [ cube
        , rotate3 (degZ 30) $ translate (0, 30, 0) $ union
          [ cube
          , rotate3 (degZ 30) $ translate (0, 30, 0) cube
          ]
        ]
```

produces:

[image]

julialongtin, 2021-02-28:
Fixed. Closing.

closed 2021-02-28 by julialongtin