Haskell-Things/ImplicitCAD #221: programs/implicitsnap.hs:182:32: error:     • Couldn't match type ‘Map String OVal’                      with ‘[ghc-prim-0.5.3:GHC.Types.Char]’

l29ah, 2019-05-12:
```
Preprocessing executable 'implicitsnap' for implicit-0.2.1..
Building executable 'implicitsnap' for implicit-0.2.1..
[1 of 1] Compiling Main             ( programs/implicitsnap.hs, dist/build/implicitsnap/implicitsnap-tmp/Main.dyn_o )

programs/implicitsnap.hs:182:32: error:
    • Couldn't match type ‘Map String OVal’
                     with ‘[ghc-prim-0.5.3:GHC.Types.Char]’
      Expected type: (String, [SymbolicObj2], [SymbolicObj3])
        Actual type: (Graphics.Implicit.ExtOpenScad.Definitions.VarLookup,
                      [SymbolicObj2], [SymbolicObj3])
    • In the first argument of ‘getWidth’, namely ‘s’
      In the expression: getWidth s
      In an equation for ‘w’: w = getWidth s
    |
182 |                 w   = getWidth s
    |                                ^
```

julialongtin, 2019-05-13:
Thanks. I'll look into this this evening.

julialongtin, 2019-05-13:
What version of GHC are you using?

l29ah, 2019-05-15:
8.6.4

julialongtin, 2019-05-20:
should be fixed. reopen if i break it again.

closed 2019-05-20 by julialongtin