Haskell-Things/ImplicitCAD #230: Toggle to enable Travis-CI build

lancelet, 2019-09-08:
Currently, there is a `.travis.yml` file in the project, but the Travis-CI build does not appear to be enabled. Doing so would allow builds of pull-requests to be built and the (currently non-operational) build badge to show the status of `master`. In turn, this could help to process PRs for the project (of which there is quite a large backlog) by confirming that they at least build correctly.

Due to ownership issues, I believe that the Travis-CI build for this project can only be "toggled on" by @colah. 

I have confirmed, in my own fork of the repo, that the Travis build works. See here for an example of a build succeeding for commit 5b45a2c: 
https://travis-ci.org/lancelet/ImplicitCAD/builds/582244741
However, although I was able to toggle CI for my own fork, as noted above, I don't think that any other contributor is able to trigger the CI build for `colah/ImplicitCAD`, aside from @colah.

Please can this be enabled?

julialongtin, 2019-09-11:
@lancelet : Colah has been absent from this project in most ways for quite some time. I'd have to have a rather serious use case to bother him.

Right now i am tracking three branches with code that may be worth integrating, but none in a state where it can 'just be added'. therefore i don't see what build checks on merge requests would get me.

I'll leave this open, as it would be nice to have this working, but it's not a priority for me.

julialongtin, 2020-12-21:
We have github actions now. closing this.

closed 2020-12-21 by julialongtin