Haskell-Things/ImplicitCAD #231: corrupt stl in example15.scad

seanjensengrey, 2019-09-08:
[example15.scad](https://github.com/colah/ImplicitCAD/blob/master/Examples/example15.scad) creates malformed output, the ring has broken scraggly edges. 

```
// $fn=100;

difference() {
    sphere(20);
    cylinder(r=17, h=100, center = true);
}
```

Rendered with OpenSCAD
[image]

rendered with `extopenscad` 

```
$ extopenscad example15.scad
Loading File.
Processing File.
Rendering 3D object from example15.scad to example15.stl with resolution 1.818181818181818 in box ((-20.0,-20.0,-20.0),(20.0,20.0,20.0))
DifferenceR3 0.0 [Sphere 20.0,Translate3 (0.0,0.0,-50.0) (ExtrudeR 0.0 (Circle 17.0) 100.0)]
```

and the resulting 

[image]

Rendered in Cura to confirm

[image]

malformed stl for reference

[example15.stl.zip](https://github.com/colah/ImplicitCAD/files/3588093/example15.stl.zip)

This is from ImplicitCAD built from src on osx.

```
$ cabal --version
cabal-install version 3.0.0.0
compiled using version 3.0.0.0 of the Cabal library
```

```
$ ghc -v
Glasgow Haskell Compiler, Version 8.6.4, stage 2 booted by GHC version 8.6.3
Using binary package database: /Users/seanj/.ghcup/ghc/8.6.4/lib/ghc-8.6.4/package.conf.d/package.cache
package flags []
loading package database /Users/seanj/.ghcup/ghc/8.6.4/lib/ghc-8.6.4/package.conf.d
wired-in package ghc-prim mapped to ghc-prim-0.5.3
wired-in package integer-gmp mapped to integer-gmp-1.0.2.0
wired-in package base mapped to base-4.12.0.0
wired-in package rts mapped to rts
wired-in package template-haskell mapped to template-haskell-2.14.0.0
wired-in package ghc mapped to ghc-8.6.4
*** Deleting temp files:
Deleting: 
*** Deleting temp dirs:
Deleting: 
ghc: no input files
Usage: For basic information, try the `--help' option.
```

I installed Haskell (8.6.4) via [ghcup](https://www.haskell.org/ghcup/) and `cabal` via `ghcup cabal-install`. This is on OSX 10.13.6 using clang/llvm from xcode. 

```
clang --version
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

lancelet, 2019-09-11:
The scraggly edges are a side-effect of the evaluation algorithm that ImplicitCAD uses. They are likely to be particularly objectionable on a model like this with sharp edges and thin features.

You can improve the output quality quite a lot by decreasing the `--resolution` parameter passed to `extopenscad` (at the expense of greater computation time and a larger mesh).

For example, if  you take it to the extreme (CAUTION: this takes **1 min 48 sec** on my machine, producing a **68 MB** STL file!):
```bash
extopenscad --resolution 0.1 example15.scad
```
[image]
The jagged edge is still there; just a lot finer.

There may be theoretical ways around this, such as using some form of gradient information, and there is some literature on the topic. For example, this paper illustrates one approach:
Battacharya, A and Wenger R (2013) Constructing Isosurfaces with Sharp Edges and Corners using Cube Merging. Computer Graphics Forum. 32(3), 11-20. [doi: 10.1111/cgf.12088](https://web.cse.ohio-state.edu/~wenger.4/papers/sharpiso.pdf) (link to author's copy)

However, I don't know if there's any current ImplicitCAD development in that direction.

julialongtin, 2019-09-11:
@lancelet : thanks for the answer.

@seanjensengrey : lancelet is right, and it's also worth noting implicitcad has a $res variable, so you can code this into your files (smaller is better!), the same way you coded in $fn in your openscad example to say "100 sides is approximately a circle".

closed 2019-09-11 by julialongtin