Haskell-Things/ImplicitCAD #232: build fails on Arch Linux:  "Polymorphic local binding with no type signature"

kjcole, 2019-09-23:
(Disclaimer: "I are not a Haskell wizard." In fact, I don't know the language at all.)

First, cabal complained that it wants everything prefixed with "new-" (update is now new-update, configure is now new-configure, install is now new-install, repl is now new-repl, bla-bla-bla is now new-bla-bla-bla).

Then it complained that ~/.cabal/store/ghc-8.6.5/package.db does not exist.

Then it complained that there are files missing in the ‘base-4.12.0.0’ package,

Specifically (after following the README, then putting "new-" in front
of everything, then searching the interwebs for info about a missing
package.db:

```
$ ghc --version
The Glorious Glasgow Haskell Compilation System, version 8.6.5

$ cabal --version
cabal-install version 2.4.0.0
compiled using version 2.4.0.1 of the Cabal library 

$  cabal new-repl -w ghc-8.6.5 --build-dep fail
...
Failed to build Boolean-0.2.4 (which is required by implicit-0.2.1)...
Failed to build NumInstances-1.4 (which is required by implicit-0.2.1)...
Failed to build blaze-svg-0.3.6.1 (which is required by implicit-0.2.1)...
Failed to build byteorder-1.0.4 (which is required by implicit-0.2.1)...
Failed to build call-stack-0.2.0 (which is required by implicit-0.2.1)...

$  cabal new-configure --disable-library-vanilla   \
                       --enable-shared             \
                       --enable-executable-dynamic \
                       --ghc-options=-dynamic
...
$ cabal new-repl -w ghc-8.6.5 --build-dep fail
...
Resolving dependencies...
Build profile: -w ghc-8.6.5 -O1
In order, the following will be built (use -v for more details):
 - implicit-0.2.1 (lib) (first run)
Configuring library for implicit-0.2.1..
Preprocessing library for implicit-0.2.1..
GHCi, version 8.6.5: http://www.haskell.org/ghc/  :? for help

<interactive>:1:1: error: [-Wmissing-local-signatures, -Werror=missing-local-signatures]
    Polymorphic local binding with no type signature:
      _compileParsedExpr :: forall a.
                            ghc-prim-0.5.3:GHC.Types.IO a -> ghc-prim-0.5.3:GHC.Types.IO a
cabal: repl failed for implicit-0.2.1.
$
```

kjcole, 2019-09-23:
I only went through all of the above because

```
$ cabal update && cabal install implicit
```
failed. However, after going through all of the above and then trying 

```
$ cabal new-update && cabal new-install implicit
```
it appears to have installed. So, still something wonky, but I could not tell you what or where.

julialongtin, 2019-12-02:
There's a new makefile to use for the new-update new-install cabal 3 stuff. closing issue.

closed 2019-12-02 by julialongtin