Haskell-Things/ImplicitCAD #235: Stack build on Windows Fails "could not find module..."

canadaduane, 2019-12-01:
I'm trying a clean install using `stack build` (Stack version 2.1.3), and all the dependencies seem to build fine, but then it results in the following errors:

```
programs\implicitsnap.hs:15:1: error:
    Could not find module `Prelude'
    Perhaps you haven't installed the "dyn" libraries for package `base-4.12.0.0'?
    Use -v to see a list of the files searched for.
   |
15 | import Prelude (IO, Maybe(Just, Nothing), String, Bool(True, False), Show, ($), (++), (>), (.), (-), (/), (*), (**), (==), null, sqrt, min, max, minimum, maximum, show, return, map, otherwise, filter, not)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

programs\implicitsnap.hs:17:1: error:
    Could not find module `Control.Applicative'
    Perhaps you haven't installed the "dyn" libraries for package `base-4.12.0.0'?
    Use -v to see a list of the files searched for.
   |
17 | import Control.Applicative ((<|>))
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

programs\implicitsnap.hs:19:1: error:
    Could not find module `Snap.Core'
    Perhaps you haven't installed the "dyn" libraries for package `snap-core-1.0.4.0'?
    Use -v to see a list of the files searched for.
   |
19 | import Snap.Core (Snap, route, writeBS, method, Method(GET), modifyResponse, setContentType, setTimeout, getRequest, rqParam)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

programs\implicitsnap.hs:20:1: error:
    Could not find module `Snap.Http.Server'
    Perhaps you haven't installed the "dyn" libraries for package `snap-server-1.1.0.0'?
    Use -v to see a list of the files searched for.
   |
20 | import Snap.Http.Server (quickHttpServe)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

programs\implicitsnap.hs:21:1: error:
    Could not find module `Snap.Util.GZip'
    Perhaps you haven't installed the "dyn" libraries for package `snap-core-1.0.4.0'?
    Use -v to see a list of the files searched for.
   |
21 | import Snap.Util.GZip (withCompression)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

programs\implicitsnap.hs:24:1: error:
    Could not find module `Graphics.Implicit'
    Perhaps you haven't installed the "dyn" libraries for package `implicit-0.2.1'?
    Use -v to see a list of the files searched for.
   |
24 | import Graphics.Implicit (runOpenscad, extrudeR)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

programs\implicitsnap.hs:27:1: error:
    Could not find module `Graphics.Implicit.ExtOpenScad.Definitions'
    Perhaps you haven't installed the "dyn" libraries for package `implicit-0.2.1'?
    Use -v to see a list of the files searched for.
   |
27 | import Graphics.Implicit.ExtOpenScad.Definitions (OVal(ONum), VarLookup, lookupVarIn, Message, Message(Message), MessageType(TextOut), ScadOpts(ScadOpts))
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

programs\implicitsnap.hs:30:1: error:
    Could not find module `Graphics.Implicit.ObjectUtil'
    Perhaps you haven't installed the "dyn" libraries for package `implicit-0.2.1'?
    Use -v to see a list of the files searched for.
   |
30 | import Graphics.Implicit.ObjectUtil (getBox2, getBox3)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

programs\implicitsnap.hs:33:1: error:
    Could not find module `Graphics.Implicit.Definitions'
    Perhaps you haven't installed the "dyn" libraries for package `implicit-0.2.1'?
    Use -v to see a list of the files searched for.
   |
33 | import Graphics.Implicit.Definitions (SymbolicObj2(UnionR2),Completed 91 action(s).
terminateProcess: permission denied (Permission denied)
```

What am I missing?

julialongtin, 2019-12-01:
I really don't know stack well, and definately don't know windows, but...

"dyn" libraries sound like something is trying to build dynamic executables, when the dependencies were not speced for that. and looking at the cabal file.. yeah, this is my fault. patch made, and pushed to master.

update, and try again. :)

... you move from chicago to canada? :)

canadaduane, 2019-12-01:
Build success! Thanks for that.

I moved to Utah, but my roots will always be Canadian! :)

closed 2019-12-01 by canadaduane