Haskell-Things/ImplicitCAD #240: OpenSCAD function creation is extremely slow

Digital-Monk, 2020-01-21:
Coming from OpenSCAD, I love extopenscad.  And I am _thrilled_ that it's getting updates and maintenance.  I know how much work that is, and it's much appreciated.

I did find something that surprised me a lot on the performance front.  Defining an OpenSCAD function (even if never used) takes the parsing time from sub-second up to a couple of minutes.  And that's even if the function is never used, so it's not an evaluation time issue.  It even happens before the entire .escad has been parsed, because I've had it take several minutes and then report a syntax error on a later line.

Here's my function, along with support variables.  Since the function does not have to be used to experience the slowdown, this code could be placed in any .escad file.  If you comment out the last line, extopenscad will process the file very quickly (so it isn't a problem with parsing functions, because pipV doesn't cause a problem).  

`pipR = 1.5;`
`PI = 3.14159265358979323846;`
`pipV = ((4/3)*PI*pipR*pipR*pipR) / 2;`
`pipV6 = 6*pipV;`
`function pipSize(numPips) = pow(((3*(2*(pipV6/numPips)))/(4*PI)), 1/3);`

Trying to narrow it down further, the following functions parse "instantaneously", so it's something about the combination of everything above...

`function pipSize(numPips) = numPips;`
`function pipSize(numPips) = pow(27, 1/3);`
`function pipSize(numPips) = pow(pipV6/numPips, 1/3);`
`function pipSize(numPips) = pow(4*PI, 1/3);`

Moreover, if I break that apart a bit and reform it into this equivalent form:

`magic = 3/(4*PI);`
`function pipSize(numPips) = pow(2*(pipV6/numPips)*magic, 1/3);`

I get "instantaneous" parsing.  So, something about depth of parentheses?

Thanks again for all the work!

lepsa, 2020-01-23:
Hi Digital-Monk,

I haven't been able to reproduce this parsing time issue in either a ghc repl or from the command line on commit e999aa48f9d13e55a4b65e58b41d526d51b267fb. All of my tests returned without noticeable delay.

One thought I have on this is that the slow down might be due to looking up the symbols in the program state, and when the above code snippets are in a file by themselves the state is small enough that it isn't a problem.

Are you able to provide the git commit or version of the code that this bug is manifesting in? Additionally can you confirm if the attached file runs slowly on your end. It is your slow code example, but with PI renamed to avoid a variable re-declaration error that my version is giving. If it runs quickly then it gives weight to the idea that this is a state issue.

Thanks, Owen
[foo.txt](https://github.com/colah/ImplicitCAD/files/4100698/foo.txt)
.

Digital-Monk, 2020-01-23:
I'm running 0.2.0 as built by `cabal install implicit`.

It does happen with that snippet.  I just timed it at it took 2 minutes at 45 seconds.

I fired up `htop` out of curiosity while it was running, and (if I read the screen correctly), extopenscad's working memory size was ~1 terabyte.  I obviously don't have anywhere near that in my machine :)

lepsa, 2020-01-23:
The 1tb of working memory is an artifact of how x86_64 GHC handles its memory allocation. In `htop` the `M_RESIDENT` column will give you a far better idea of the memory actually in use.

Digital-Monk, 2020-01-23:
Ah.  That makes more sense.  About 11kB.  But it's burning a CPU core at 99.7% working on that 11kB...  Oh well, stale/ancient build, so kind of a moot point.  Sounds like it's already fixed, whatever it was.

closed 2020-01-23 by Digital-Monk