Haskell-Things/ImplicitCAD #241: OpenSCAD variable/parameter typos seem to be silent

Digital-Monk, 2020-01-21:
If my OpenSCAD code has typos, extopenscad seems to silently fail on the affected expression/statement.  This has been a problem when I messed up capitalization on a variable, and also when I tried to use the 'd' diameter parameter in cylinders and spheres (I guess only the 'r' radius parameter is supported for now).  OpenSCAD would also continue on with the code, but it issued warnings about the unrecognized variables/parameters.

(Hopefully I'm not just being an idiot and somehow hiding these warnings)

julialongtin, 2020-01-22:
Mind passing me some more concrete examples?

Digital-Monk, 2020-01-22:
Here's the simplest.  Note the failure to capitalize properly in the sphere usage.

`camelCase = 10;`
`sphere(r = camelcase);`

extopenscad will say "no objects to render", but won't mention that 'camelcase' is an unrecognized symbol.

lepsa, 2020-01-23:
Hi Digital-Monk, I've been unable to reproduce this in commit e999aa48f9d13e55a4b65e58b41d526d51b267fb.
Do you know which version/commit of ImplicitCAD you are running?

Output with bad casing.
```
[nix-shell:~/git/ImplicitCAD]$ cat bar.escad 
camelCase = 10;
sphere(r = camelcase);

[nix-shell:~/git/ImplicitCAD]$ cabal new-run extopenscad -- bar.escad 
Up to date
Loading File.
Processing File.
No objects to render.
Error at line 2, column 1: Variable camelcase not in scope
Error at line 2, column 1: arg Undefined not compatible with Symbol "r"
```

Output with good casing
```
[nix-shell:~/git/ImplicitCAD]$ cat bar.escad 
camelCase = 10;
sphere(r = camelCase);

[nix-shell:~/git/ImplicitCAD]$ cabal new-run extopenscad -- bar.escad 
Up to date
Loading File.
Processing File.
Rendering 3D object from bar.escad to bar.stl with resolution 0.909090909090909 in box ((-10.0,-10.0,-10.0),(10.0,10.0,10.0))
Sphere 10.0

[nix-shell:~/git/ImplicitCAD]$ cat bar.stl 
<snipped due to length and unicode replacement chartacters>
```

Digital-Monk, 2020-01-23:
I'm running whatever 0.2.0 comes out of `cabal install implicit`...  Definitely not the same as what you're running, because the build I've got puts "In box [...]" on a line by itself instead of at the end of the "Rendering ..." line.

Sorry, I'm not very familiar with haskell or cabal...

lepsa, 2020-01-23:
No worries.
That version of implicit is rather old, coming up to two years in May! We should really update the version on hackage or update the readme to let people know where/how to get a newer copy. I don't have access to do that, but Julia certainly does.

Digital-Monk, 2020-01-23:
Just curious:  What version is running for the online demo on implicitcad.org?

lepsa, 2020-01-23:
I do not know offhand, and a few minutes of digging didn't turn up anything. However, it does handle the casing and function declaration issues.

Digital-Monk, 2020-01-23:
Sounds like this issue is definitely closed, though :)  Ancient history

closed 2020-01-23 by Digital-Monk