Haskell-Things/ImplicitCAD #242: OpenSCAD cylinder and sphere don't accept diameter (d, d1, d2) parameters

Digital-Monk, 2020-01-21:
Low priority, because it's easy enough to specify radius instead.  But some kind of warning output would be useful.  For me, spheres just failed to generate anything, and cylinders used a very small radius.

julialongtin, 2020-01-22:
I'll take a look at this as well. thank you for the report!

julialongtin, 2020-01-26:
The newer version should have warned you, but.. I've added these, as well. give them a shot!

Digital-Monk, 2020-01-26:
I'm probably messing up something with the build process...  Did a `git pull`, and now `git log` shows that I'm at  8cb3a93a23541, which should be the newest.

I tried `cabal install`, which seemed to work, but afterwards extopenscad still didn't recognize `d` for `sphere()` or `d1` and `d2` for `cylinder`.

Assumed I did something wrong with cabal, erased the extopenscad binary, tried `cabal build` (huge log, several warnings, no errors) followed by `cabal install`.  Same result.

Sorry to be such a noob with cabal, but should I be doing something else to build with the new code I pulled?

julialongtin, 2020-01-26:
whoops. typo. try now. :D

Digital-Monk, 2020-01-27:
Looks good!  And cool that there's a new release!  This is so much nicer than trying to manually fillet things in OpenSCAD...  (I know there's a LOT more to ImplicitCAD than that, but that's the immediate killer feature for my simplistic mechanical designs)

closed 2020-01-27 by Digital-Monk