Haskell-Things/ImplicitCAD #244: warnings stopping build (e999aa4)

Digital-Monk, 2020-01-23:
Cloned the repo, installed dependencies, configured and tried to install.  `cabal install` dumped this:

```
cabal: Entering directory '.'
Configuring implicit-0.2.1...
Building implicit-0.2.1...
Preprocessing library implicit-0.2.1...

<no location info>: error: 
Failing due to -Werror.

on the commandline: warning:
    unrecognised warning flag: -Wmissing-export-lists

on the commandline: warning:
    unrecognised warning flag: -Wmissing-home-modules
cabal: Leaving directory '.'
```

Guessing this is caused by a missing tool or library on my end...  I do have 0.2.0 installed as well, but that shouldn't cause compilation warnings.

julialongtin, 2020-01-23:
What version of haskell is this? We've been testing on 8.4 and 8.6..

Digital-Monk, 2020-01-23:
Oh, FFS...  GHC 8.0.2...

I'm on Linux Mint 19.3 (newest) which is based on Ubuntu 18.04 (new-ish, and latest LTS)...  You'd think that its ghc package might not be ancient...

Huh.  Went to GHC home, and the only Debian-based instructions are to get it from the repos, which is what left me stuck with 8.0.2  (well, technically the GHC website instructions say to get the ghc6 package, which suggests to me how stale that page may be)

I see that GHC is up to 8.8.2, just not sure how much of a headache it's going to be to get it.  Not especially looking forward to a source build and possible dependency nightmare.

Debian packages says that their stable repo has 8.4.4.  Ubuntu package search shows that they've limited it to 8.0.2 in bionic (18.04 LTS).  Time for me to get used to adjusting my repo sources to something newer and hopefully not hose my dependencies ;)

So, once again, this is nothing on your project at all.  May want to tweak your install instructions, because unfortunately `sudo apt install ghc` can be way behind on common distributions...

Digital-Monk, 2020-01-23:
For anyone else having similar problems, the "Generic Linux" installer seems pretty smooth:  [https://www.haskell.org/platform/linux.html#linux-generic](url).  Gave me GHC 8.6.5 and cabal-install 3.0.0.0.  It also updated my command path, saving a minor step.

Followed the `git clone ...` and `cabal ...` instructions from here, and only had two issues:
1. Needed to specify an installdir.  I used /home/myname/.cabal/bin, because that was where the old stable release put itself.
2. Because I had the old stable release there, I also needed to overwrite old versions.

So, my final install command was:
`cabal install --installdir=/home/myname/.cabal/bin --overwrite-policy=always`

Your mileage may vary...

closed 2020-01-25 by julialongtin