Haskell-Things/ImplicitCAD #245: [e999aa4] Parameter matching generates errors, but works anyway

Digital-Monk, 2020-01-23:
GHC 8.6.5, extopenscad 0.2.1, commit e999aa4.

The following line in my code:
`cube(size = [80, 33, 145], r = 2);`

Generates the following on stderr:
```
Error at line 8, column 25: no instance of cube found to match given parameters.
Instances available:
Module cube[  (x=..., y=..., z=..., center) {},  (size=..., center) {} ]
Error at line 8, column 25: Insufficient parameters. Passed 2 named parameters. Couldn't match 3 parameters: "x", "y" and "z". Had 2 extra parameters. They are:"size" and "r".
Error at line 8, column 25: Too many parameters: 2 extra. Passed 2 named parameters. Couldn't match 3 parameters: "x", "y" and "z". Had 2 extra parameters. They are:"size" and "r".
Error at line 8, column 25: Too many parameters: 1 extra. Passed 2 named parameters. Had one extra parameter: "r"
```

Oddly enough, the model is built exactly as requested.  This seems to be some extra pattern matcher that is out of sync with ImplicitCAD (I notice that `r` is never recognized as a valid parameter in these messages).  I have noticed that if I don't use any named parameters and do specify all of the parameters, I don't get these "errors".  For instance, `cube([80, 33, 145], false, 2);` doesn't cause any signature errors.

Regardless, the correct model is built, and the correct object is output.

julialongtin, 2020-01-25:
Fixed, try the latest version. :)

Digital-Monk, 2020-01-26:
Looks good!  Thanks!

closed 2020-01-26 by Digital-Monk