Haskell-Things/ImplicitCAD #247: ESCAD "module" lexing/parsing can't handle a newline before closing parenthesis

nmz787, 2020-02-24:
```
module my_sphere(diam
                 ) {
    sphere(d=diam);
}
my_sphere(100);
```

when run with `extopenscad test.escad` produces:
```
Loading File.
Processing File.
No objects to render.
Error at line 3, column 5: Variable diam not in scope
Error at line 3, column 5: arg Undefined not compatible with Symbol "d"
```

Simply removing the newline before the closing parenthesis fixes the errors:
```
module my_sphere(diam){
    sphere(d=diam);
}
my_sphere(100);
```
with the output being:
```
Loading File.
Processing File.
Rendering 3D object from test_recursion.escad to test_recursion.stl with resolution 4.545454545454544 in box ((-50.0,-50.0,-50.0),(50.0,50.0,50.0))
Sphere 50.0
```

regardless of the placement of the curly-brace, i.e.:
```
module my_sphere(diam)
{
    sphere(d=diam);
}
my_sphere(100);
```
or even
 ```
module my_sphere(diam)
{sphere(d=diam);}
my_sphere(100);
```

julialongtin, 2020-03-24:
Nice catch. :)

julialongtin, 2020-04-10:
fixed, along with several other spacing issues. thanks!

closed 2020-04-10 by julialongtin