Haskell-Things/ImplicitCAD #251: Setting up dev environment

skvamme, 2020-02-28:
Following documentation for setting up dev environment failed.
OSX Catalina 10.15.3

> Prelude> :l Graphics.Implicit
[ 1 of 48] Compiling Graphics.Implicit.ExtOpenScad.Parser.Lexer ( Graphics/Implicit/ExtOpenScad/Parser/Lexer.hs, interpreted )

Graphics/Implicit/ExtOpenScad/Parser/Lexer.hs:14:1: error:
    Could not find module ‘Control.Monad.Identity’
    Perhaps you meant
      Control.Monad.Identity (from mtl-2.2.2)
      Control.Monad.Cont (from mtl-2.2.2)
      Control.Monad.State (from mtl-2.2.2)
   |
14 | import "monads-tf" Control.Monad.Identity (Identity)
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Failed, no modules loaded.

>

julialongtin, 2020-03-01:
We just merged a PR related to this; try again from the newest git?

skvamme, 2020-03-01:
Came a little bit longer with the newest git. But stopped with another module it could't find:

`[ 4 of 48] Compiling Graphics.Implicit.Definitions ( Graphics/Implicit/Definitions.hs, interpreted )
*** Parser [Graphics.Implicit.Definitions]:
!!! Parser [Graphics.Implicit.Definitions]: finished in 2.14 milliseconds, allocated 6.220 megabytes
*** Renamer/typechecker [Graphics.Implicit.Definitions]:
!!! Renamer/typechecker [Graphics.Implicit.Definitions]: finished in 0.15 milliseconds, allocated 0.192 megabytes

Graphics/Implicit/Definitions.hs:83:1: error:
    Could not find module ‘Data.VectorSpace’
    Locations searched:
      Data/VectorSpace.hs
      Data/VectorSpace.lhs
      Data/VectorSpace.hsig
      Data/VectorSpace.lhsig
   |
83 | import Data.VectorSpace (Scalar, InnerSpace, (<.>))
   | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Upsweep partially successful.
*** Deleting temp files:
Deleting: 
Failed, three modules loaded.
*Graphics.Implicit.IntegralUtil> 
Leaving GHCi.
`

skvamme, 2020-03-02:
Solved
The versions I have (ghc 8.6.5 and cabal 3.0.0.0) does not play nice together. I had to use the v1-versions of configure and install. (v2- is default in cabal 3.0.0.0).
The makefile works fine for creating the binaries as it is, but when starting ghci there was things missing in the environment file. Here's the issue https://github.com/haskell/cabal/issues/5559

closed 2020-03-02 by skvamme