Haskell-Things/ImplicitCAD #254: Wanted: Haskel source for CAD

skvamme, 2020-04-04:
It's easy to find ExtOpenScad source, but I am a developer in functional languages since 30 years so I want Haskel source to look at when I am programming. (I always use my old programs as a reference). I'm new to Haskel and new to ImplicitCAD so I wonder if someone could be kind and send me some old ImplicitCAD Haskel CAD objects, nothing fancy, just the basics. s@kvamme.se

If I get some code maybe I could improve the Haskel-part of the documentation :)
Thanks!

lancelet, 2020-04-05:
All the files in the `Examples` folder with `.hs` extensions are Haskell examples that you can execute to produce either STL (3D) or SVG (2D) files. For example:
https://github.com/colah/ImplicitCAD/blob/master/Examples/example13.hs

I hope that helps.

skvamme, 2020-04-05:
Yes, thanks :) done that. I was curious if there are more Haskell coded graphics around. I would like to 3D-print a Toroid shaped plastic gasket but did not find so much to lean on in the docs. ImplicitCAD has been around for a long time, so I recon there is quite a large user base who have produced a lot of code. /S