Haskell-Things/ImplicitCAD #262: Installation fails on macOS

wbrickner, 2020-05-10:
Hello,

I followed the instructions for release and development installation on macOS, and both fail.

Here's the output I get when installing the release version from cabal:

```
wbrickner@mbp ~ % cabal update && cabal install implicit
Downloading the latest package list from hackage.haskell.org
To revert to previous state run:
    cabal v2-update 'hackage.haskell.org,2020-05-10T19:09:09Z'
Resolving dependencies...
Build profile: -w ghc-8.10.1 -O1
In order, the following will be built (use -v for more details):
 - implicit-0.0.5 (lib:implicit, exe:extopenscad) (requires build)
Starting     implicit-0.0.5 (all, legacy fallback)
Building     implicit-0.0.5 (all, legacy fallback)

Failed to build implicit-0.0.5.
Build log ( /Users/wbrickner/.cabal/logs/ghc-8.10.1/mplct-0.0.5-9172cdee.log
):
Configuring implicit-0.0.5...
Warning: 'ghc-options: -threaded' has no effect for libraries. It should only
be used for executables.
Warning: 'ghc-options: -rtsopts' has no effect for libraries. It should only
be used for executables.
Preprocessing executable 'extopenscad' for implicit-0.0.5..
Building executable 'extopenscad' for implicit-0.0.5..

Graphics/Implicit/ExtOpenScad/Util/OVal.hs:1:14: warning:
    -XOverlappingInstances is deprecated: instead use per-instance pragmas OVERLAPPING/OVERLAPPABLE/OVERLAPS
  |
1 | {-# LANGUAGE OverlappingInstances, ViewPatterns, RankNTypes, ScopedTypeVariables, TypeSynonymInstances, FlexibleInstances #-}
  |              ^^^^^^^^^^^^^^^^^^^^

<no location info>: warning: [-Wmissing-home-modules]
    These modules are needed for compilation but not listed in your .cabal file's other-modules: 
        Graphics.Implicit
        Graphics.Implicit.Definitions
        Graphics.Implicit.Export
        Graphics.Implicit.Export.Definitions
        Graphics.Implicit.Export.MarchingSquares
        Graphics.Implicit.Export.MarchingSquaresFill
        Graphics.Implicit.Export.NormedTriangleMeshFormats
        Graphics.Implicit.Export.PolylineFormats
        Graphics.Implicit.Export.RayTrace
        Graphics.Implicit.Export.Render
        Graphics.Implicit.Export.Render.Definitions
        Graphics.Implicit.Export.Render.GetLoops
        Graphics.Implicit.Export.Render.GetSegs
        Graphics.Implicit.Export.Render.HandlePolylines
        Graphics.Implicit.Export.Render.HandleSquares
        Graphics.Implicit.Export.Render.Interpolate
        Graphics.Implicit.Export.Render.RefineSegs
        Graphics.Implicit.Export.Render.TesselateLoops
        Graphics.Implicit.Export.Symbolic.Rebound2
        Graphics.Implicit.Export.Symbolic.Rebound3
        Graphics.Implicit.Export.SymbolicFormats
        Graphics.Implicit.Export.SymbolicObj2
        Graphics.Implicit.Export.SymbolicObj3
        Graphics.Implicit.Export.TextBuilderUtils
        Graphics.Implicit.Export.TriangleMeshFormats
        Graphics.Implicit.Export.Util
        Graphics.Implicit.ExtOpenScad
        Graphics.Implicit.ExtOpenScad.Default
        Graphics.Implicit.ExtOpenScad.Definitions
        Graphics.Implicit.ExtOpenScad.Eval.Expr
        Graphics.Implicit.ExtOpenScad.Eval.Statement
        Graphics.Implicit.ExtOpenScad.Parser.Expr
        Graphics.Implicit.ExtOpenScad.Parser.Statement
        Graphics.Implicit.ExtOpenScad.Parser.Util
        Graphics.Implicit.ExtOpenScad.Primitives
        Graphics.Implicit.ExtOpenScad.Util.ArgParser
        Graphics.Implicit.ExtOpenScad.Util.OVal
        Graphics.Implicit.ExtOpenScad.Util.StateC
        Graphics.Implicit.MathUtil
        Graphics.Implicit.ObjectUtil
        Graphics.Implicit.ObjectUtil.GetBox2
        Graphics.Implicit.ObjectUtil.GetBox3
        Graphics.Implicit.ObjectUtil.GetImplicit2
        Graphics.Implicit.ObjectUtil.GetImplicit3
        Graphics.Implicit.Primitives
[ 1 of 46] Compiling Graphics.Implicit.Definitions ( Graphics/Implicit/Definitions.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Definitions.o )
[ 2 of 46] Compiling Graphics.Implicit.Export.Definitions ( Graphics/Implicit/Export/Definitions.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/Definitions.o )
[ 3 of 46] Compiling Graphics.Implicit.Export.MarchingSquares ( Graphics/Implicit/Export/MarchingSquares.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/MarchingSquares.o )
[ 4 of 46] Compiling Graphics.Implicit.Export.MarchingSquaresFill ( Graphics/Implicit/Export/MarchingSquaresFill.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/MarchingSquaresFill.o )
[ 5 of 46] Compiling Graphics.Implicit.Export.Render.Definitions ( Graphics/Implicit/Export/Render/Definitions.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/Render/Definitions.o )
[ 6 of 46] Compiling Graphics.Implicit.Export.Render.GetLoops ( Graphics/Implicit/Export/Render/GetLoops.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/Render/GetLoops.o )
[ 7 of 46] Compiling Graphics.Implicit.Export.Render.HandlePolylines ( Graphics/Implicit/Export/Render/HandlePolylines.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/Render/HandlePolylines.o )
[ 8 of 46] Compiling Graphics.Implicit.Export.Render.HandleSquares ( Graphics/Implicit/Export/Render/HandleSquares.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/Render/HandleSquares.o )
[ 9 of 46] Compiling Graphics.Implicit.Export.Render.Interpolate ( Graphics/Implicit/Export/Render/Interpolate.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/Render/Interpolate.o )
[10 of 46] Compiling Graphics.Implicit.Export.Symbolic.Rebound2 ( Graphics/Implicit/Export/Symbolic/Rebound2.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/Symbolic/Rebound2.o )
[11 of 46] Compiling Graphics.Implicit.Export.Symbolic.Rebound3 ( Graphics/Implicit/Export/Symbolic/Rebound3.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/Symbolic/Rebound3.o )
[12 of 46] Compiling Graphics.Implicit.Export.TextBuilderUtils ( Graphics/Implicit/Export/TextBuilderUtils.hs, dist/build/extopenscad/extopenscad-tmp/Graphics/Implicit/Export/TextBuilderUtils.o )

Graphics/Implicit/Export/TextBuilderUtils.hs:20:6: error:
    Ambiguous occurrence ‘<>’
    It could refer to
       either ‘Prelude.<>’,
              imported from ‘Prelude’ at Graphics/Implicit/Export/TextBuilderUtils.hs:5:8-48
              (and originally defined in ‘GHC.Base’)
           or ‘Graphics.Implicit.Export.TextBuilderUtils.<>’,
              defined at Graphics/Implicit/Export/TextBuilderUtils.hs:72:1
   |
20 |     ,(<>)
   |      ^^^^
cabal: Failed to build implicit-0.0.5. See the build log above for details.
```

Any advice would be appreciated.

wbrickner, 2020-05-10:
Machine: `MacBook Pro (13-inch, 2018, Four Thunderbolt 3 Ports)`
OS: `macOS Catalina 10.15.4 (19E287)`

Running freshly-installed `cabal` from `brew`.

wbrickner, 2020-05-10:
FYI development version also fails, it's a version conflict between the requirements of `snap-server` and `implicit`:

```
$ clear && cabal install --only-dependencies --minimize-conflict-set
```

```
cabal: Could not resolve dependencies:
[__0] trying: implicit-0.3.0.1 (user goal)
[__1] trying: snap-server-1.1.1.1 (dependency of implicit)
[__2] trying: network-3.1.1.1 (dependency of snap-server)
[__3] next goal: base (dependency of implicit)
[__3] rejecting: base-4.14.0.0/installed-4.14.0.0 (conflict: snap-server =>
base>=4.6 && <4.14)
[__3] skipping: base-4.14.0.0 (has the same characteristics that caused the
previous version to fail: excluded by constraint '>=4.6 && <4.14' from
'snap-server')
[__3] rejecting: base-4.13.0.0, base-4.12.0.0, base-4.11.1.0, base-4.11.0.0,
base-4.10.1.0, base-4.10.0.0, base-4.9.1.0, base-4.9.0.0, base-4.8.2.0,
base-4.8.1.0, base-4.8.0.0, base-4.7.0.2, base-4.7.0.1, base-4.7.0.0,
base-4.6.0.1, base-4.6.0.0, base-4.5.1.0, base-4.5.0.0, base-4.4.1.0,
base-4.4.0.0, base-4.3.1.0, base-4.3.0.0, base-4.2.0.2, base-4.2.0.1,
base-4.2.0.0, base-4.1.0.0, base-4.0.0.0, base-3.0.3.2, base-3.0.3.1
(constraint from non-upgradeable package requires installed instance)
[__3] fail (backjumping, conflict set: base, implicit, snap-server)
After searching the rest of the dependency tree exhaustively, these were the
goals I've had most trouble fulfilling: base, snap-server, network, unix,
implicit
```

(I'm a Haskell noob, so package management / conflict resolution is a dark art with which I am not yet familiar).

skvamme, 2020-05-10:
I installed on Catalina 10.15.3 after a little bit struggle with incompatible versions, see Setting up dev environment #251 if it makes any sense.

wbrickner, 2020-05-10:
It starts to work but I still get similar issues. Am I doing something wrong?

Environment details:

`$ ghc --version` /  `The Glorious Glasgow Haskell Compilation System, version 8.10.1`

`$ cabal --version` / `cabal-install version 3.2.0.0
compiled using version 3.2.0.0 of the Cabal library`

My release installation attempt:

```
$ cabal v1-install implicit
```

```
Resolving dependencies...
Starting     Boolean-0.2.4
Starting     NumInstances-1.4
Starting     byteorder-1.0.4
Starting     colour-2.3.5
Starting     newtype-generics-0.5.4
Starting     parallel-3.2.2.0
Starting     blaze-builder-0.4.1.0
Starting     hashable-1.3.0.0
Building     NumInstances-1.4
Building     Boolean-0.2.4
Building     byteorder-1.0.4
Building     parallel-3.2.2.0
Building     newtype-generics-0.5.4
Building     colour-2.3.5
Building     hashable-1.3.0.0
Building     blaze-builder-0.4.1.0
Completed    byteorder-1.0.4
Starting     primitive-0.7.0.1
Completed    newtype-generics-0.5.4
Starting     transformers-compat-0.6.5
Completed    Boolean-0.2.4
Starting     zlib-0.6.2.1
Building     primitive-0.7.0.1
Completed    parallel-3.2.2.0
Starting     storable-endian-0.2.6
Building     transformers-compat-0.6.5
Building     zlib-0.6.2.1
Building     storable-endian-0.2.6
Completed    hashable-1.3.0.0
Starting     MemoTrie-0.6.10
Completed    blaze-builder-0.4.1.0
Starting     unordered-containers-0.2.10.0
Building     MemoTrie-0.6.10
Completed    NumInstances-1.4
Starting     blaze-markup-0.8.2.5
Completed    transformers-compat-0.6.5
Building     unordered-containers-0.2.10.0
Completed    storable-endian-0.2.6
Building     blaze-markup-0.8.2.5
Completed    colour-2.3.5
Starting     ansi-terminal-0.10.3
Completed    MemoTrie-0.6.10
Starting     vector-space-0.16
Building     ansi-terminal-0.10.3
Building     vector-space-0.16
Completed    zlib-0.6.2.1
Completed    ansi-terminal-0.10.3
Starting     ansi-wl-pprint-0.6.9
Completed    primitive-0.7.0.1
Starting     vector-0.12.1.2
Building     ansi-wl-pprint-0.6.9
Building     vector-0.12.1.2
Completed    vector-space-0.16
[... 162 lines cut ...]
array==0.5.4.0/installed-0.5.4.0, snap-server => array>=0.2 && <0.4)
[_13] skipping: snap-server-0.5.1.4, snap-server-0.5.1.3, snap-server-0.5.1.2,
snap-server-0.5.1.1, snap-server-0.5.1, snap-server-0.5.0, snap-server-0.4.3,
snap-server-0.4.2, snap-server-0.4.1, snap-server-0.4.0.2,
snap-server-0.4.0.1, snap-server-0.4.0, snap-server-0.3.1.1,
snap-server-0.3.1, snap-server-0.3.0, snap-server-0.2.16.2,
snap-server-0.2.16.1, snap-server-0.2.16, snap-server-0.2.15.1,
snap-server-0.2.15, snap-server-0.2.14.1, snap-server-0.2.14,
snap-server-0.2.13.3, snap-server-0.2.13.2, snap-server-0.2.13.1,
snap-server-0.2.13, snap-server-0.2.12, snap-server-0.2.11,
snap-server-0.2.10.2, snap-server-0.2.10.1, snap-server-0.2.10,
snap-server-0.2.9, snap-server-0.2.8.1, snap-server-0.2.8,
snap-server-0.2.7.2, snap-server-0.2.7.1, snap-server-0.2.7,
snap-server-0.2.6, snap-server-0.2.5, snap-server-0.2.4, snap-server-0.2.2,
snap-server-0.2.1, snap-server-0.1.4, snap-server-0.1.3, snap-server-0.1.2,
snap-server-0.1.1 (has the same characteristics that caused the previous
version to fail: excludes 'array' version 0.5.4.0)
[_13] fail (backjumping, conflict set: array, implicit, primitive,
snap-server, time)
After searching the rest of the dependency tree exhaustively, these were the
goals I've had most trouble fulfilling: base, snap-server, time, statistics,
template-haskell, vector, primitive, pretty, unix, criterion, network,
vector-th-unbox, math-functions, implicit, array, monad-par, vector-algorithms
Try running with --minimize-conflict-set to improve the error message.
```

skvamme, 2020-05-10:
Try cabal v1-configure and then v1-install

wbrickner, 2020-05-10:
```
$ cabal v1-configure && cabal v1-install --only-dependencies
```

Output (identical with/without `--only-dependencies` flag):
```
Resolving dependencies...
Warning: solver failed to find a solution:
Could not resolve dependencies:
[__0] trying: implicit-0.3.0.1 (user goal)
[__1] unknown package: criterion (dependency of implicit)
[__1] fail (backjumping, conflict set: criterion, implicit)
After searching the rest of the dependency tree exhaustively, these were the
goals I've had most trouble fulfilling: implicit, criterion
Trying configure anyway.
Configuring implicit-0.3.0.1...
cabal: Encountered missing or private dependencies:
aeson -any,
criterion -any,
hspec -any,
snap-core -any,
snap-server -any,
utf8-string -any
```

skvamme, 2020-05-10:
My mistake, v1-install should come before v1-configure (Im new to Haskell)

wbrickner, 2020-05-10:
Both command sequences fail, regardless of order, same output as before (first command will fail, so the output will be alternate depending on order).

skvamme, 2020-05-10:
Have you tried to run make ? I think I cleaned it all up between my failed installations and when it suddenly worked out.

wbrickner, 2020-05-10:
Both `$ make clean && cabal v1-install --only-dependencies && cabal v1-configure` and `$ make && cabal v1-install --only-dependencies && cabal v1-configure` provide the following 
 output (_looks_ identical to previous errors):

```
Resolving dependencies...
cabal: Could not resolve dependencies:
[__0] trying: implicit-0.3.0.1 (user goal)
[__1] trying: criterion-1.5.6.2 (dependency of implicit)
[__2] trying: vector-algorithms-0.8.0.3 (dependency of criterion)
[__3] trying: primitive-0.7.0.1/installed-2k3g9qX0zz16vEv34R307m (dependency
of vector-algorithms)
[__4] trying: vector-0.12.1.2/installed-7XbwjHOLjQXGIwimXzS4vB (dependency of
criterion)
[__5] trying: time-1.9.3/installed-1.9.3 (dependency of criterion)
[__6] trying: statistics-0.15.2.0 (dependency of criterion)
[__7] trying: vector-th-unbox-0.2.1.7 (dependency of statistics)
[__8] trying: template-haskell-2.16.0.0/installed-2.16.0.0 (dependency of
vector-th-unbox)
[__9] trying: pretty-1.1.3.6/installed-1.1.3.6 (dependency of
template-haskell)
[_10] trying: monad-par-0.3.5 (dependency of statistics)
[_11] trying: array-0.5.4.0/installed-0.5.4.0 (dependency of monad-par)
[_12] trying: math-functions-0.3.3.0 (dependency of statistics)
[_13] next goal: snap-server (dependency of implicit)
[_13] rejecting: snap-server-1.1.1.1 (conflict: primitive =>
base==4.14.0.0/installed-4.14.0.0, snap-server => base>=4.6 && <4.14)
[_13] rejecting: snap-server-1.1.1.0 (conflict: primitive =>
base==4.14.0.0/installed-4.14.0.0, snap-server => base>=4.8 && <4.13)
[_13] rejecting: snap-server-1.1.0.0 (conflict: primitive =>
base==4.14.0.0/installed-4.14.0.0, snap-server => base>=4.6 && <4.13)
[_13] rejecting: snap-server-1.0.3.3, snap-server-1.0.3.2,
snap-server-1.0.3.1, snap-server-1.0.3.0, snap-server-1.0.2.2,
snap-server-1.0.2.1, snap-server-1.0.2.0, snap-server-1.0.1.1 (conflict:
primitive => base==4.14.0.0/installed-4.14.0.0, snap-server => base>=4 &&
<4.11)
[_13] rejecting: snap-server-1.0.1.0 (conflict: time==1.9.3/installed-1.9.3,
snap-server => time>=1.0 && <1.7)
[_13] skipping: snap-server-1.0.0.0, snap-server-0.9.5.1, snap-server-0.9.5.0,
snap-server-0.9.4.6, snap-server-0.9.4.5, snap-server-0.9.4.4,
snap-server-0.9.4.3, snap-server-0.9.4.2, snap-server-0.9.4.1,
snap-server-0.9.4.0, snap-server-0.9.3.4, snap-server-0.9.3.3,
snap-server-0.9.3.1, snap-server-0.9.3, snap-server-0.9.2.4,
snap-server-0.9.2.3, snap-server-0.9.2.2, snap-server-0.9.2.1,
snap-server-0.9.2, snap-server-0.9.0, snap-server-0.8.1.1, snap-server-0.8.1,
snap-server-0.8.0.1, snap-server-0.8.0, snap-server-0.7.0.1, snap-server-0.7,
snap-server-0.6.0.1, snap-server-0.6.0, snap-server-0.5.5, snap-server-0.5.4,
snap-server-0.5.3.1, snap-server-0.5.3 (has the same characteristics that
caused the previous version to fail: excludes 'time' version 1.9.3)
[_13] rejecting: snap-server-0.5.2 (conflict:
array==0.5.4.0/installed-0.5.4.0, snap-server => array>=0.2 && <0.4)
[_13] skipping: snap-server-0.5.1.4, snap-server-0.5.1.3, snap-server-0.5.1.2,
snap-server-0.5.1.1, snap-server-0.5.1, snap-server-0.5.0, snap-server-0.4.3,
snap-server-0.4.2, snap-server-0.4.1, snap-server-0.4.0.2,
snap-server-0.4.0.1, snap-server-0.4.0, snap-server-0.3.1.1,
snap-server-0.3.1, snap-server-0.3.0, snap-server-0.2.16.2,
snap-server-0.2.16.1, snap-server-0.2.16, snap-server-0.2.15.1,
snap-server-0.2.15, snap-server-0.2.14.1, snap-server-0.2.14,
snap-server-0.2.13.3, snap-server-0.2.13.2, snap-server-0.2.13.1,
snap-server-0.2.13, snap-server-0.2.12, snap-server-0.2.11,
snap-server-0.2.10.2, snap-server-0.2.10.1, snap-server-0.2.10,
snap-server-0.2.9, snap-server-0.2.8.1, snap-server-0.2.8,
snap-server-0.2.7.2, snap-server-0.2.7.1, snap-server-0.2.7,
snap-server-0.2.6, snap-server-0.2.5, snap-server-0.2.4, snap-server-0.2.2,
snap-server-0.2.1, snap-server-0.1.4, snap-server-0.1.3, snap-server-0.1.2,
snap-server-0.1.1 (has the same characteristics that caused the previous
version to fail: excludes 'array' version 0.5.4.0)
[_13] fail (backjumping, conflict set: array, implicit, primitive,
snap-server, time)
After searching the rest of the dependency tree exhaustively, these were the
goals I've had most trouble fulfilling: base, snap-server, time, statistics,
template-haskell, vector, primitive, pretty, unix, criterion, network,
vector-th-unbox, math-functions, implicit, array, monad-par, vector-algorithms
Try running with --minimize-conflict-set to improve the error message.
```

wbrickner, 2020-05-10:
(this is the latest development version btw)

skvamme, 2020-05-10:
Yes, we do not have identical software versions but gut feeling is that this is a really Haskell thing. It's midnight here now so I have to go, if something pops up I will write here.

wbrickner, 2020-05-10:
Another clue: even installing the dependencies listed on [ImplicitCAD](http://www.implicitcad.org/docs/install) site fails:

```
$ cabal update && cabal install hashmap parallel parsec plugins JuicyPixels blaze-builder
```
(as well as the `v1-install` variant) yields:

```
Downloading the latest package list from hackage.haskell.org
To revert to previous state run:
    cabal v2-update 'hackage.haskell.org,2020-05-10T20:36:56Z'
cabal: Could not resolve dependencies:
[__0] trying: implicit-0.3.0.1 (user goal)
[__1] trying: criterion-1.5.6.2 (dependency of implicit)
[__2] trying: vector-0.12.1.2 (dependency of criterion)
[__3] trying: primitive-0.7.0.1 (dependency of vector)
[__4] trying: time-1.9.3/installed-1.9.3 (dependency of criterion)
[__5] trying: statistics-0.15.2.0 (dependency of criterion)
[__6] trying: vector-th-unbox-0.2.1.7 (dependency of statistics)
[__7] trying: template-haskell-2.16.0.0/installed-2.16.0.0 (dependency of
vector-th-unbox)
[__8] trying: pretty-1.1.3.6/installed-1.1.3.6 (dependency of
template-haskell)
[__9] trying: monad-par-0.3.5 (dependency of statistics)
[_10] trying: array-0.5.4.0/installed-0.5.4.0 (dependency of monad-par)
[_11] trying: math-functions-0.3.3.0 (dependency of statistics)
[_12] next goal: snap-server (dependency of implicit)
[_12] rejecting: snap-server-1.1.1.1 (conflict: time =>
base==4.14.0.0/installed-4.14.0.0, snap-server => base>=4.6 && <4.14)
[_12] rejecting: snap-server-1.1.1.0 (conflict: time =>
base==4.14.0.0/installed-4.14.0.0, snap-server => base>=4.8 && <4.13)
[_12] rejecting: snap-server-1.1.0.0 (conflict: time =>
base==4.14.0.0/installed-4.14.0.0, snap-server => base>=4.6 && <4.13)
[_12] rejecting: snap-server-1.0.3.3, snap-server-1.0.3.2,
snap-server-1.0.3.1, snap-server-1.0.3.0, snap-server-1.0.2.2,
snap-server-1.0.2.1, snap-server-1.0.2.0, snap-server-1.0.1.1 (conflict: time
=> base==4.14.0.0/installed-4.14.0.0, snap-server => base>=4 && <4.11)
[_12] rejecting: snap-server-1.0.1.0 (conflict: time==1.9.3/installed-1.9.3,
snap-server => time>=1.0 && <1.7)
[_12] skipping: snap-server-1.0.0.0, snap-server-0.9.5.1, snap-server-0.9.5.0,
snap-server-0.9.4.6, snap-server-0.9.4.5, snap-server-0.9.4.4,
snap-server-0.9.4.3, snap-server-0.9.4.2, snap-server-0.9.4.1,
snap-server-0.9.4.0, snap-server-0.9.3.4, snap-server-0.9.3.3,
snap-server-0.9.3.1, snap-server-0.9.3, snap-server-0.9.2.4,
snap-server-0.9.2.3, snap-server-0.9.2.2, snap-server-0.9.2.1,
snap-server-0.9.2, snap-server-0.9.0, snap-server-0.8.1.1, snap-server-0.8.1,
snap-server-0.8.0.1, snap-server-0.8.0, snap-server-0.7.0.1, snap-server-0.7,
snap-server-0.6.0.1, snap-server-0.6.0, snap-server-0.5.5, snap-server-0.5.4,
snap-server-0.5.3.1, snap-server-0.5.3 (has the same characteristics that
caused the previous version to fail: excludes 'time' version 1.9.3)
[_12] rejecting: snap-server-0.5.2 (conflict:
array==0.5.4.0/installed-0.5.4.0, snap-server => array>=0.2 && <0.4)
[_12] skipping: snap-server-0.5.1.4, snap-server-0.5.1.3, snap-server-0.5.1.2,
snap-server-0.5.1.1, snap-server-0.5.1, snap-server-0.5.0, snap-server-0.4.3,
snap-server-0.4.2, snap-server-0.4.1, snap-server-0.4.0.2,
snap-server-0.4.0.1, snap-server-0.4.0, snap-server-0.3.1.1,
snap-server-0.3.1, snap-server-0.3.0, snap-server-0.2.16.2,
snap-server-0.2.16.1, snap-server-0.2.16, snap-server-0.2.15.1,
snap-server-0.2.15, snap-server-0.2.14.1, snap-server-0.2.14,
snap-server-0.2.13.3, snap-server-0.2.13.2, snap-server-0.2.13.1,
snap-server-0.2.13, snap-server-0.2.12, snap-server-0.2.11,
snap-server-0.2.10.2, snap-server-0.2.10.1, snap-server-0.2.10,
snap-server-0.2.9, snap-server-0.2.8.1, snap-server-0.2.8,
snap-server-0.2.7.2, snap-server-0.2.7.1, snap-server-0.2.7,
snap-server-0.2.6, snap-server-0.2.5, snap-server-0.2.4, snap-server-0.2.2,
snap-server-0.2.1, snap-server-0.1.4, snap-server-0.1.3, snap-server-0.1.2,
snap-server-0.1.1 (has the same characteristics that caused the previous
version to fail: excludes 'array' version 0.5.4.0)
[_12] fail (backjumping, conflict set: array, implicit, snap-server, time)
After searching the rest of the dependency tree exhaustively, these were the
goals I've had most trouble fulfilling: base, snap-server, time, network,
unix, statistics, criterion, template-haskell, pretty, vector-th-unbox, array,
primitive, vector, implicit, math-functions, monad-par
Try running with --minimize-conflict-set to improve the error message.
```

skvamme, 2020-05-11:
Investigating, running make, and it was ok except for lots of warnings: "Could not specialise imported function."

Did git pull to get latest ImplicitCAD.

Running make again, and it didn't compile. Tried make clean and make, same negative.

julialongtin, 2020-05-11:
have you tried using stack? I suspect a lot of what you're having troubles with is ghc 8.10 issues... i don't have 8.10 here to try.

skvamme, 2020-05-11:
ghc --version                        
The Glorious Glasgow Haskell Compilation System, version 8.6.5

I use ghcup -v install

skvamme, 2020-05-11:
Upgrading cabal: ghcup install-cabal 3.2.0.0

Changing OS=linux in Makefile to OS=osx

Solved!

wbrickner, 2020-05-15:
That worked.  Changed the line in makefile.

closed 2020-05-15 by wbrickner