Haskell-Things/ImplicitCAD #267: Parallelization?

wbrickner, 2020-06-08:
Hello,

(I believe) there exist highly parallel marching cubes implementations, some of which may utilize the GPU.

Is this something that's feasible to integrate with ImplicitCAD?
I'd like to generate extremely large models at high resolution (for the purposes of additive manufacturing / procedural generation), but unfortunately (I'd expect) ImplicitCAD to take tens of hours for the types of models I'm interested in.

If this is outside the project scope I understand, but in its place might we add a progress bar?

Also, I've been thinking of reimplementing an implicit description model similar to that of ImplicitCAD, potentially in Rust.  There are a few old blog posts from the original author explaining the fundamentals which have proven invaluable.  If I have questions I hope it's OK to reach out over another channel.

julialongtin, 2020-06-08:
add +RTS -N -qg to the end of the command line. this is a very parallel engine, you just have to know how to unlock it. :)

julialongtin, 2020-06-08:
also, there's a lot of speedup that hasn't been done. issue #265 is a good place to start, or just actually fixing issue #98 would unlock a lot of speed.

wbrickner, 2021-03-02:
Needed to look this back up.  Is there a compelling reason not to enable this parallelism by default?

julialongtin, 2021-03-02:
I believe it is enabled by default, in the version in github.

julialongtin, 2021-03-02:
also, reading back on your initial request, it may be worth noting that the server ImplicitCAD.org uses to render is backed by 28 cores, and 256G of ram. feel free to (ab)use.

wbrickner, 2021-03-02:
Oh my. I'm waiting for the cubes to stop marching right now, so yes I may do that.

Is there any way to set the resolution on the web interface? I always use the command line flag.

julialongtin, 2021-03-02:
$RES