Haskell-Things/ImplicitCAD #268: linear_extrude should support XY-scaling

mietek, 2020-06-09:
It would be nice if `linear_extrude` accepted a `scale : ℝ → (ℝ, ℝ)` argument that would compute the scale factor for each of the two XY-dimensions separately.

FlyingGaz, 2020-06-10:
One question that remains to be solved is the type of the scale parameter in `extrudeRM` and the constructor `ExtrudeRM` (Haskell API).

Right now the scale parameter is `Either ℝ (ℝ -> ℝ)`, but going forward we need 3 variants:
- constant `ℝ`
- constant `ℝ2`
- function `ℝ -> Either ℝ ℝ2`

I came up with two possible solutions:
- `Either (Either ℝ ℝ2) (ℝ -> Either ℝ ℝ2)`
- `data ExtrudeScale = C1 ℝ | C2 ℝ2 | F (ℝ -> Either ℝ ℝ2)` (or a generalized name in case we want to reuse the type)

Let me know, what you think and I'll make a proper pull request.

mietek, 2020-06-10:
I would suggest fully unnesting the nested sum type:
* `data ExtrudeScale = C1 ℝ | C2 ℝ2 | F1 (ℝ -> ℝ) | F2 (ℝ -> ℝ2)`

FlyingGaz, 2020-06-10:
Unnesting the function is currently not possible, because the extopenscad parser doesn't evaluate the type of a function. Type errors are delayed until [runtime](https://github.com/colah/ImplicitCAD/blob/dd53cd13270f3da3bdeff221ae28ca6bf2d0119e/Graphics/Implicit/ExtOpenScad/Util/OVal.hs#L103).

mietek, 2020-06-10:
Well, then I would go with your `data ExtrudeScale`.

PR #269 (2020-06-12): Support XY-scaling on linear_extrude (resolves #268)
This pull request resolves #268.

The scale parameter of `linear_extrude` now supports `(ℝ, ℝ)` and `ℝ -> (ℝ, ℝ)` in addition to `ℝ` and `ℝ -> ℝ`. This allows `x` and `y` to be scaled independently from each other while extruding.

Example demonstrating all 4 variants:
```
union() {
    linear_extrude(height = 10, scale = 2)
    square(10, center = true);

    translate([30, 0, 0])
    linear_extrude(height = 10, scale(z) = 1 + z / 10)
    square(10, center = true);
    
    translate([0, 30, 0])
    linear_extrude(height = 10, scale = [2, 1 [... cut]
commit 75d447cc (2020-06-12): Support XY-scaling on linear_extrude (resolves #268)
commit 9ac59e95 (2020-06-12): Merge pull request #269 from FlyingGaz/issue268

Support XY-scaling on linear_extrude (resolves #268)
closed 2020-06-12 by julialongtin