Haskell-Things/ImplicitCAD #274: OpenSCAD parser issue

Jacoby6000, 2020-07-10:
I've come across an openscad lib which seems to find bugs in the parser. 

https://www.dkprojects.net/openscad-threads/threads.scad 

```
extopenscad lib/threads.scad 
Loading File.
Processing File.
(line 100, column 8):
unexpected "m"
expecting "//", "/*" or "("
```

The line(s) referenced:
```
module metric_thread (diameter=8, pitch=1, length=1, internal=false, n_starts=1,
                      thread_size=-1, groove=false, square=false, rectangle=0,
                      angle=30, taper=0, leadin=0, leadfac=1.0, test=false)
```

julialongtin, 2020-07-12:
sadly, that can be a problem inside of the code in the module, not just at the line it is reported.

julialongtin, 2020-07-12:
looking at that scad file, there are several primitives we do not support.

Jacoby6000, 2020-07-14:
Ah I see. I'll see if I can do some hacking. 👍