Haskell-Things/ImplicitCAD #276: [OpenSCAD compat] Include relative paths are evaluated differently

Depau, 2020-08-18:
Hi,

I was trying to rebuild [this project](https://github.com/Depau/aegis-mini-skin) (not the best code, I admit) with `extopenscad` with `--fopenscad-compat` and I noticed that relative imports are evaluated differently.

Specifically, OpenSCAD evaluates relative paths starting from the location of the SCAD file.

To reproduce: https://github.com/Depau/aegis-mini-skin; run `make SCAD='./path/to/extopenscad -O'`:

```
$ make SCADC='/usr/bin/extopenscad -O'
/usr/bin/extopenscad -O -o stl/stubs/decoration_side.stl stubs/decoration_side.scad
Loading File.
Processing File.
extopenscad: /tmp/aegis-mini-skin/../config.scad: openFile: does not exist (No such file or directory)
```

⇒ OpenSCAD would load `config.scad` from `/tmp/aegis-mini-skin/stubs/../config.scad`

---

Side note: someone made an Arch Linux AUR package for ImplicitCAD, if you wanna mention it in the website :) https://aur.archlinux.org/packages/implicitcad/

julialongtin, 2020-08-24:
I believe it's fair to call this broken. :)

lepsa, 2024-04-18:
I've made a fix for this that should resolve the issue.