Haskell-Things/ImplicitCAD #278: Download Rendering on website broken

hannesweisbach, 2020-10-17:
I tried to render & download the following example from the website (http://www.implicitcad.org/examples/round-plus#) as SVG, but I got only a file containing:

```
var Shape = function(){
var s = this;
THREE.Geometry.call(this);
function vec(x,y,z){return new THREE.Vector3(x,y,z);}
function v(x,y,z){s.vertices.push(vec(x,y,z));}
function f(a,b,c){s.faces.push(new THREE.Face3(a,b,c));}
v(-5.855059e-1,-8.782588e0,3.565074e-1);
v(-5.855059e-1,-9.110735e0,4.8387095e-1);
[…]
```

So … JavaScript code? Definitely not SVG ;)

julialongtin, 2021-02-27:
reproduced. what the heck? ;)

julialongtin, 2021-02-27:
ok, fixed. also fixed .OBJ and .DXF exporting, while i was at it.

closed 2021-02-27 by julialongtin