Haskell-Things/ImplicitCAD #279: Summary for curated list of code CAD

Irev-Dev, 2020-10-31:
Hey Christopher 👋 ,

I finally got around to putting together a list of all of the code-CAD projects that I know of [here](https://kurthutten.com/blog/curated-code-cad), and also on [Github](https://github.com/Irev-Dev/curated-code-cad).

I've put a short summary of each one, but it's hard to do them all justice. for ImplicitCAD I've only got:
Inspired by OpenScad with a very similar language, implemented in Haskell.

If you would like to give me a summary of the project, and perhaps some points of how it differs from other similar projects would be happy to include it. I think max three paragraphs would be about right. Open a PR on reply to this issue and I'll include it.

julialongtin, 2020-10-31:
Looks reasonably good. ImplicitCAD starts with a capital I, includes the ability to write definitions in Haskell, instead of just OpenSCAD, and is part of an 'almost stack' of tools including ExplicitCAD (for a GUI), and HSlice (for an STL slicer).

Irev-Dev, 2020-10-31:
Thanks @julialongtin, I've updated that.
Let me know if you want it updated again at any time.

closed 2020-10-31 by Irev-Dev

Irev-Dev, 2021-01-01:
Me again.

I also want to add some of the best real-world examples of Code-CAD being used out in the wild. Do you know of any good projects using ImplicitCAD that I can feature?

So far what I've found is:
- [Acedemics using it for reactor design](https://github.com/ukaea/paramak)
- [Lego compatible blocks](https://fxbricks.com/)
- [specialised in-house equipment for drones](https://twitter.com/robsmallshire/status/1343477584458346497)
- [Pruse 3d printer](https://github.com/prusa3d/Original-Prusa-i3)