Haskell-Things/ImplicitCAD #283: Bug: unexpected result when union()ing with radius

hannesweisbach, 2020-11-02:
Disclaimer: I haven't read the text about the theory behind ImplicitCAD. The described "bug" might very well be the result that math dictates. Nonetheless it is not was I was expecting. The issue appears when two bodies of the same height are unioned with a radius. Some sort of "blob" is added to the top and bottom, when I expected those to be flat.

Offending minimal (non) working example:
```
union() {
  // does not work
  union(r = 2) {
    cube(10);
    cylinder(d= 10, h = 10);
  }

  // this is what I expected
  translate([20, 0])
  linear_extrude(10)
  union(r = 2) {
    square(10);
    circle(d = 10);
  }
}
```

Rendered result:

[image]

As already stated, I expected the rounded union on the left to have a smooth top and bottom, since they are the same height. IMO there is no need to add rounding there or even enlarge the bodies! In this case I was able to work around the issue by using something like on the left: union()ing shapes in 2D and then extruding them to height. This only adds fillets where I would expect them.

As I said in the beginning, I'm not sure that this is a bug or intended behaviour, but I didn't expect that result.

Oh, and decreasing `$res` didn't make the "humps" go away, it only made them smoother, but kept them at the same height.

julialongtin, 2020-11-07:
yeah, this is a natural effect of unioning in 3d. the 'hump' is space where you are R distance from the sphere and the cube.

so this, as far as i can tell, is just how the math works. you're wanting to union in 2D, so unioning in 2d and extruding is the right approach.

tl;dr: this is just how math works. won't fix.

l29ah, 2021-06-28:
Duplicate of https://github.com/Haskell-Things/ImplicitCAD/issues/204

closed 2021-07-02 by hannesweisbach