Haskell-Things/ImplicitCAD #285: Main module missing haddock

isovector, 2020-11-05:
As a newcomer to CAD (but not to Haskell), the haddock pages leave much to be desired. For example, what is this `R` parameter to all of the `write` functions? It's completely undocumented. Diving into the source suggests it's some sort of resolution, but I'm unsure if I should be giving it big or small values.

isovector, 2020-11-08:
Here are some things I've learned in playing around with the system.

1. Smaller resolution values result in clearer models (and superlinear slow downs when rendering)
2. The camera for PNG3 appears to be facing forwards
3. The coordinate system is `(width, depth, height)`
4. Functions that end in `R` have a "roundness" as their first parameter. It's unclear what units this is measured in.
5. Size and position units appear to be in `mm`

julialongtin, 2020-11-12:
Could you submit those as changes to the comments? the state of the haddock docs is sad, my apologies.

isovector, 2020-11-12:
Gladly. I'll try to get that done this weekend.

julialongtin, 2020-11-15:
I believe you've fixed this. ;)