Haskell-Things/ImplicitCAD #288: RotateExtrude should use radians not degrees

raptortech-js, 2020-11-15:
Most of ImplicitCAD uses Radians, but the `RotateExtrude` implementation uses degrees:
https://github.com/colah/ImplicitCAD/blob/5ddd75006cc8f1d6a0e184f56c3714a9097ce6b0/Graphics/Implicit/ObjectUtil/GetImplicit3.hs#L170-L176

This is a breaking change, but I think we should change the API to use radians not degrees. Fwiw, I use `RotateExtrude` to implement a screw, and I calculate a number of rotations anyways and then have to convert to degrees for this function.

Note that I'm mostly thinking about the Haskell eDSL side, I have no familiarity with ExtOpenScad

julialongtin, 2020-11-19:
The reason this uses degrees is that in the scad language, the call to this accepts degrees. in order to change this and maintain openscad compatibility, you would have to put a special case in place.

sorki, 2020-11-20:
I've stumbled on this as well and think we should change the Haskell interface but keep the extopenscad one compatible not to break existing code/examples.

PR #299 (2020-12-26): Make rotateExtrude operate in radians
Before, rotateExtrude operated over degrees for compatability reasons
with scad. This is an odd choice for the haskell library itself, so
instead we make rotateExtrude do the right thing, and put a shim into
openscad to make the transformation.

Fixes #288
closed 2020-12-26 by julialongtin