Haskell-Things/ImplicitCAD #289: Any interest in combinators?

isovector, 2020-11-17:
One of Haskell's greatest advantages is the extent to which purity allows for code-reuse. Today ImplicitCAD has support for a few primitives, but nothing really that builds on top of them. 

For example, when aligning sub-objects, it can be extremely helpful to modify where their origin lies. My projects make excessive use of 

```haskell
-- | Move an object's origin to be flush against the bottom
slam :: SymbolicObj3 -> SymbolicObj3
slam obj =
  let (_, _, z) = getOrigin obj
   in translate (mk3 0 0 (-z)) obj

getOrigin :: Object obj vec => obj -> vec
getOrigin = fst . getBox
```

as another example, I have a combinator which symmetrically places objects in polar coordinates around a central axis.

Is there any interest in merging combinators of this sort, or should I put them into a new repository?

julialongtin, 2020-11-19:
no idea. neither of those look especially useful to me, but i just write in scad. mind providing an example of their utility?

sorki, 2020-11-20:
As a casual user of Haskell interface I'm interested and also in other usability improvements of it like `cubeR`.

What do you think about adding short-hands for e.g. `cube = cubeR 0` as well?

isovector, 2020-11-20:
@sorki I'm in. It seems to me that the rounding operation should be its own combinator --- though I don't know how feasible this is today.

isovector, 2020-11-20:
I wrote a proposal for that in #298

closed 2020-12-06 by isovector