Haskell-Things/ImplicitCAD #315: Cabal flag for utils

juhp, 2020-11-27:
Just a small packaging suggestion: how about adding a build flag to make the building of `Benchmark` optional (or allowing it not to be built)?
Maybe `implicitsnap` and `docgen` too perhaps if it makes sense? I am not sure how they are used i general.

Thanks for this interesting project.

sorki, 2020-11-27:
Both `implicitsnap` and `docgen` are behind respective flags since #272.

You can disable benchmark too with `cabal configure --disable-benchmarks` which creates `cabal.project.local`. Not sure if this is needed for development though - like you can just not run benchmark. Benchmark was also switched from executable to benchmark type in that PR so it's not part of default set of executables anymore.

juhp, 2020-11-28:
Okay, cool - sorry I should have checked more carefully. I think I installed from Hackage.

closed 2020-11-28 by juhp