Haskell-Things/ImplicitCAD #323: Monoid instance is FUBAR

isovector, 2020-12-01:
There are no primitives that work as `mempty`. `cube 0` still tracks distance from the origin, and `union []`'s implicit is equal to `0` everywhere.

isovector, 2020-12-01:
Also `union` isn't associative for some reason

isovector, 2020-12-01:
```
  1) monoid laws (getImplicit) (a <> b) <> c = a <> (b <> c)                                                         
       Falsified (after 21 tests and 15 shrinks):
         Rotate3 (Quaternion 0.9960149774672645 (V3 4.6856813189826194e-2 (-1.8194877999712162e-2) (-7.367190871121673e-2))) (ExtrudeR 0.29527 (SquareR 2.442 (-3.3827657897387633,-1.0858261587348177)) 9.71752)
         ExtrudeR 0.0 (IntersectR2 5.10065 [SquareR 0.56644 (0.0,0.0),Translate2 (-0.0,-0.0) (SquareR 0.26109 (0.0,0.0)),SquareR 0.30417 (0.0,0.0),SquareR 0.24298 (0.0,0.0),Circle 0.0,PolygonR 0.82068 [(0.0,0.0),(0.0,0.0),(0.0,0.0),(0.0,0.0),(0.0,0.0)],Circle 0.0,SquareR 0.97443 (0.0,0.0),Circle 0.0,PolygonR 0.43754 [(0.0,0.0),(0.0,0.0),(0.0,0.0),(0.0,0.0)]]) 1.0
         UnionR3 0.0 [Sphere 0.1]
         ((0.0,0.0,4.0),())
         3.9 /= 3.53015
```

cartazio, 2020-12-02:
Sounds like it’s a good thing semigroups are a thing then?

isovector, 2020-12-02:
But it's not even associative ;)

cartazio, 2020-12-02:
Is there any way to recover associativity?

isovector, 2020-12-02:
My guess is that the non-associativity is a bug. There's just no way in hell that union wouldn't be associative. As for getting the monoid out, I've started work on new primitives to do just this in #324.

PR #324 (2020-12-04): Fix monoid instances; add tests of algebraic properties for symbolic objects
This PR:

1. Primarily, fixes #323, via:
2. Adding two new primitives to `Object`: `emptySpace` and `fullSpace`. The former acts as a proper unit for union, and the latter is its corresponding dual (and also annihilator for union)
3. Enforces associativity of the `(<>)` operation by fiat. The underlying bug in `union` is still present, but `(<>)` correctly reassociates everything so it Just Works
4. Adds tests to prove that the monoid instances are lawful
5. Adds thousands(!) of other tests, proving tons of algebraic  [... cut]
closed 2020-12-04 by julialongtin