Haskell-Things/ImplicitCAD #353: Tesselation is broken

isovector, 2020-12-15:
Filing a new bug to act as a little notepad for myself.

The following object has a huge erroneous spike when rendered with resolution=2:

```haskell
translate (V3 0.0 (-27.5) 0.0) (union [translate (V3 62.0 0.0 33.0) (union [translate (V3 (-124.0) 0.0 0.0) (cubeR 0.0 False (V3 124.0 124.0 2.0)),translate (V3 (-56.0) 62.0 (-33.0)) (union [translate (V3 0.0 0.0 35.0) (union [translate (V3 0.0 0.0 (-17.5)) (difference (translate (V3 0.0 0.0 (-17.5)) (cylinder 42.0 35.0)) [translate (V3 0.0 0.0 (-15.5)) (cylinder 40.0 33.0)]),union [translate (V3 0.0 0.0 (-2.0)) (cylinder 48.0 2.0),emptySpace]]),emptySpace])]),translate (V3 (-3.5) (-50.0) 0.0) (union [translate (V3 0.0 (-9.5) 0.0) (union [translate (V3 0.0 0.0 8.5) (rotate3 (3.141592653589793,0.0,0.0) (translate (V3 (-62.0) (-9.5) (-8.5)) (cubeR 0.0 False (V3 124.0 19.0 17.0)))),union [translate (V3 61.0 0.0 0.0) (union [translate (V3 0.0 (-4.0) 0.0) (cubeR 0.0 False (V3 8.0 8.0 2.0)),translate (V3 (-2.0) (-4.0) 0.0) (cubeR 0.0 False (V3 2.0 8.0 8.0))]),emptySpace]]),translate (V3 0.0 52.0 0.0) (union [translate (V3 (-52.0) (-54.0) 0.0) (cubeR 0.0 False (V3 104.0 54.0 17.0)),translate (V3 0.0 (-2.0) 0.0) (difference (extrudeR 0.0 (polygonR 0.0 [V2 (-48.0) 0.0,V2 48.0 0.0,V2 0.0 60.0]) 17.0) [translate (V3 0.0 0.0 2.0) (extrudeR 0.0 (polygonR 0.0 [V2 (-46.0) 0.0,V2 46.0 0.0,V2 0.0 58.0]) 13.0)])])])])
```

[image]

isovector, 2020-12-15:
The STL contains the following triangles, which have X and Y components 10x bigger than anything else in the file. Presumably these are the culprits:

```stl
facet normal 7.808541e-1 -6.247135e-1 0.0e0
outer loop
vertex -3.200224e1 -3.1188211e0 -1.3e1
vertex 1.1662979e2 1.8266225e2 -1.3e1
vertex 4.2313778e1 8.977171e1 -1.2e1
endloop
endfacet
facet normal 7.808541e-1 -6.247135e-1 0.0e0
outer loop
vertex 1.1662979e2 1.8266225e2 -1.3e1
vertex 1.1662979e2 1.8266225e2 -1.1e1
vertex 4.2313778e1 8.977171e1 -1.2e1
endloop
endfacet
facet normal 7.808541e-1 -6.247135e-1 0.0e0
outer loop
vertex 1.1662979e2 1.8266225e2 -1.1e1
vertex -3.200224e1 -3.1188211e0 -1.1e1
vertex 4.2313778e1 8.977171e1 -1.2e1
endloop
endfacet
facet normal -7.846233e-1 6.1997277e-1 0.0e0
outer loop
vertex 1.1662979e2 1.8266225e2 -1.1e1
vertex 1.1662979e2 1.8266225e2 -1.3e1
vertex 4.289338e1 8.934312e1 -1.2e1
endloop
endfacet
facet normal -7.846233e-1 6.1997277e-1 0.0e0
outer loop
vertex 1.1662979e2 1.8266225e2 -1.3e1
vertex -3.0843039e1 -3.9760044e0 -1.3e1
vertex 4.289338e1 8.934312e1 -1.2e1
endloop
endfacet
facet normal -7.846233e-1 6.1997277e-1 0.0e0
outer loop
vertex -3.0843039e1 -3.9760044e0 -1.3e1
vertex -3.0843039e1 -3.9760044e0 -1.1e1
vertex 4.289338e1 8.934312e1 -1.2e1
endloop
endfacet
facet normal -7.846233e-1 6.1997277e-1 0.0e0
outer loop
vertex -3.0843039e1 -3.9760044e0 -1.1e1
vertex 1.1662979e2 1.8266225e2 -1.1e1
vertex 4.289338e1 8.934312e1 -1.2e1
endloop
endfacet
```

isovector, 2020-12-15:
The spike from another angle. It's not just a single triangle:

[image]

isovector, 2020-12-15:
Rendering the same object with resolutions `{3,4,5,6,9,10}` also produce crazy spikes. It doesn't (visibly) occur at any of `{1, 2.5, 7, 8}`.

isovector, 2020-12-15:
At resolution 10:

[image]

The vertex here is at `vertex -3.4342197e3 1.4210855e-14 -1.6341694e1`, which is 10x further away from the model than in the resolution 2 case. Maybe the resolution has something to how crazy these triangles get? If so, it would explain why we don't see them when we decrease the resolution.

isovector, 2020-12-15:
The spike remains all the way until resolution `10.0695652173913030580365557398181408`. At `10.0695652173913030580365557398181409`, we get a drastically different mesh:

[image]

isovector, 2020-12-15:
I instrumented the triangle generation with this:

```haskell
mkTriangle :: (ℝ3, ℝ3, ℝ3) -> Triangle
mkTriangle (a, b, c)  =
  if abs a >= 3400
     then markDebugger $ Triangle (a, b, c)
     else Triangle (a, b, c)

markDebugger :: Triangle -> Triangle
markDebugger = id
```

and then added a breakpoint to `markDebugger`. The resulting stack trace is this:

```
> :hist
-1  : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:140:11-43)
-2  : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:139:6-10)
-3  : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:139:6-18)
-4  : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:(139,3)-(141,28))
-5  : tesselateLoop (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:113:29-48)
-6  : rnf (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Definitions.hs:192:28-38)
-7  : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:141:11-28)
-8  : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:139:6-10)
-9  : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:139:6-18)
-10 : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:(139,3)-(141,28))
-11 : tesselateLoop (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:113:29-48)
-12 : rnf (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Definitions.hs:192:28-38)
-13 : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:141:11-28)
-14 : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:139:6-10)
-15 : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:139:6-18)
-16 : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:(139,3)-(141,28))
-17 : tesselateLoop (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:113:29-48)
-18 : rnf (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Definitions.hs:192:28-38)
-19 : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:141:11-28)
-20 : mkTriangle (/home/sandy/prj/ImplicitCAD/Graphics/Implicit/Export/Render/TesselateLoops.hs:139:6-10)
```

isovector, 2020-12-15:
This is the offending line:

https://github.com/colah/ImplicitCAD/blob/f04ebb57a7809067a36cf97f223891d49babdb4f/Graphics/Implicit/Export/Render/TesselateLoops.hs#L93

isovector, 2020-12-15:
some printf of the values in scope (ghci doesn't track this for some stupid reason)

```haskell
mid= V3 (-318.298453912407) (-5.033029043219576) 2.365568688888259
mid'= V3 (-65.49999999494523) 24165.67241446491 155.07559049902787
midval= 252.798453912407
deriv= 1.045809707875378e-2
normal= V3 (-1.0458097078962893e-2) (-0.9999253558637543) (-6.317507913012856e-3)
```

isovector, 2020-12-15:
The weird triangle comes from `path`, which is defined:

https://github.com/colah/ImplicitCAD/blob/f04ebb57a7809067a36cf97f223891d49babdb4f/Graphics/Implicit/Export/Render/TesselateLoops.hs#L78

so maybe it's `shrinkLoop` that's broken

sorki, 2020-12-15:
Maybe simpler repro:

```haskell
res = 2
s = 6.79
obj = rotate3 (V3 0 (0.9 * 2 * pi) 0) $ union $ [
    translate (V3 0 0 (-s)) (cylinder2 1 3 7) 
  , translate (V3 0 0 s) $ rotate3 (V3 pi 0 0) $  cylinder2 1 3 7
  ]  
```

[image]

With `res = 1` and others it's fine
[image]

isovector, 2020-12-16:
Found it! https://github.com/colah/ImplicitCAD/pull/350/commits/e7b2cbd8f420801f1268d429a6f33b1208a4c1df

`interpolate` doesn't do a very good job of finding roots.

isovector, 2020-12-16:
https://github.com/colah/ImplicitCAD/pull/355 is a better step here, but it's still not right

PR #355 (2020-12-17): Fix the broken triangle generation
This fixes the broken triangles. Detailing was attempting to do... something... and it didn't work.

All of the changed golden tests pass my eyeball test.

Fixes #353
closed 2020-12-17 by julialongtin