Haskell-Things/ImplicitCAD #384: Can't read stl file

hasufell, 2021-03-25:
File: https://github.com/dpapavas/lagrange-keyboard/blob/master/things/right.stl

```
$ extopenscad -O things/right.scad 
Loading File.
Processing File.
No objects to render.
SyntaxError at line 1, column 8:
unexpected "("
expecting "="
<<ghc: 1221144 bytes, 1 GCs, 200224/200224 avg/max bytes residency (1 samples), 13M in use, 0.004 INIT (0.004 elapsed), 0.002 MUT (0.008 elapsed), 0.000 GC (0.001 elapsed) :ghc>>
```

julialongtin, 2021-03-25:
so, implicitcad takes scad, and generates stl, not the other way arround. :)

hasufell, 2021-03-25:
> so, implicitcad takes scad, and generates stl, not the other way arround. :)

Yes, see the log:

```
$ extopenscad -O things/right.scad 
```

Merely the link is incorrect (the scad file needs to be generated). I uploaded the file here: https://gist.github.com/hasufell/360c81482d0a1f7a7d49c46b42968fff

julialongtin, 2021-03-26:
ok, looking at that scad file, I can tell you that implicitcad has no color operation, and worse for your model, has no hull operation. determining how to implement hull in a signed distance field is a hard problem.

julialongtin, 2021-04-02:
what scad generator are you using?

hasufell, 2021-04-02:
It's generated from clojure code: https://github.com/dpapavas/lagrange-keyboard/blob/master/src/lagrange_keyboard/core.clj

It appears to be [scad-clj "0.5.4-SNAPSHOT"](https://github.com/dpapavas/lagrange-keyboard/blob/7c54cbfd058b9b525e36a7992c7d30c54e0a7f68/project.clj#L7).

julialongtin, 2021-04-02:
https://github.com/farrellm/scad-clj/issues/38

julialongtin, 2021-04-02:
Started a dialog with upstream to see if we can fix this integration for you.

julialongtin, 2021-04-03:
so, it looks like the library is not the source of the issue, but the decisions of the keyboard's designers.
switching which upstream i bother. :)